nplan/HomeButtons #5: Possibility to run on wall charger?

nt7, 2022-11-11:
Hi!
Huge fan of your product. 
I was wondering if I could permanently attach a usb C cable to the device and power it continuously via 5 (?) Volt?
Best, n

nplan, 2022-11-11:
Thanks! 😃 

Unfortunately powering it through USB-C while the battery is missing is **not working** at the moment (HW revision 2.1). It may appear to be working if you try it, but it can't handle the current peaks when Wi-Fi is communicating.

If the battery is inserted, keeping it connected is also not recommended, since it would keep the battery fully charged at all times.

We are working on a fix for this. Could you maybe share what is your use case and what is the reason you would prefer USB-C power over battery?

nplan, 2022-11-18:
Wrote a short [guide](https://docs.home-buttons.com/hardware_hacks/#usb-power-without-battery) for enabling USB-C only power. Requires some soldering.

closed 2022-11-18 by nplan

nt7, 2022-11-22:
Thanks!
My main usecase will be permanentlynear an AC outlet, so there's no reason to power via battery. 
The hack would be perfectly suited!
Now I just have to wait until the product is back on Tindie, as I missed buying it when you answered ;(