nplan/HomeButtons #6: Live information on screen

zoic21, 2022-11-11:
Hello,
Did you plan to add the capacity of live information on screen ? To preserve battery device can connect every 10 min to Mqtt and read retain topic with information to display (home mode, ext temp, alarm mode...)

Thank in advance for your reply

nplan, 2022-11-11:
Currently that is not implemented. But I'm open to suggestions what to add in the future.

As I see it, the main problem with displaying other info on the screen is that it covers button labels, which are the main feature of the device.

I was thinking of maybe adding the possibility of displaying some data at the top and then only have labels for the bottom 4 buttons.

What do you think about this?

zoic21, 2022-11-12:
Hello,

It can be amazing and very useful, with this two line we can display information like home mode and ext temp.

zoic21, 2022-11-13:
Hello,
I got an idea for this request. Maybe it's possible to use "scroll" information for top of device. For exemple we can send this retain topic on mqtt : 
```
[
{"line_1" : "Mode","line_2" : "Night"},
{"line_1" : "Ext","line_2" : "17°C"},
{"line_1" : "Int","line_2" : "21°C"}
]
```

And on home buttons we can used 2 first button to scroll between information (for exemple : Mode Night then Ext 17°C then Int 21°C), other 4 button leave like actually.

I don't realize how difficult it is to code this, but I think it can be usefull

nplan, 2022-11-17:
Will add something like this, thanks for your feedback!

MiCkSoftware, 2022-11-22:
Hi all,
Another idea on the same topic : if there was a rest API or MQTT topic to update buttons label it would then be possible to change display on button click and open a lot of possibilities (like using two buttons to scroll trough multiple screens, ...)
Cheers !

EarMaster, 2022-11-23:
The ability to change the labels without entering setup mode would be a great addition. This would allow a lot more flexible usage. Of course battery life will go down significantly when there are regular updates to the screen, but this would be something everybody has to decide for themselves.

nplan, 2022-11-23:
Sure, changing labels via MQTT is definitely on the list. The new label message can be posted with retain flag and the label will change in 10 minutes at the latest, when it wakes up for sensor reporting. Or right after a button press.

I'll also make the 10 minute wake up interval user adjustable.

AndersNielsenUFST, 2022-12-05:
Thumbs up on this one. I am thinking of a scenario where you click "Activate Alarm" and the display text changes to "Windows are open".

Hamsterman, 2022-12-05:
Thumbs on this one. I am thinking of a scenario where you click "Activate Alarm" and the text changes to "Windows are open"

zoic21, 2022-12-27:
Hello,
I just update to last firmware, it's perfectly work !!!! Thank a lot for this update

closed 2022-12-27 by zoic21