nplan/HomeButtons #9: Question (not an issue) - Have you considered double-click functionality?

Koendejongh, 2022-11-19:
Hi,

Is it possible to add double-click functionality to home buttons?

Thanks,
Koen

nplan, 2022-11-19:
No, not at the moment. It's on the list of future improvements.

nplan, 2023-02-02:
Added this feature in v2.0.1-beta. It works slightly differently than what @gon0 suggested to be compatible with Home Assistant. There is now a separate topic and a separate trigger entity for single, double, triple and quadruple press. The payload is still "PRESS" in all cases. He LEDs flash the number of presses. Hold functionality is not implemented since it's used for controlling other features of Home Buttons.

closed 2023-02-02 by nplan

MatthiasLohr, 2023-02-24:
> The LEDs flash the number of presses

Can't confirm that.

Single: LED just on
Double: LED just on
Triple: LED on - off - on
Quadruple: LED on - off - on - off - on

nplan, 2023-02-24:
In sleep mode the last blink always holds until device goes back to sleep. In awake mode all blinks are the same length.

Is it actually registering the correct number of presses and just blinking wrong?

You might be clicking too quickly. See issue #26.

MatthiasLohr, 2023-02-24:
> You might be clicking too quickly. See issue #26.

Oh, yes! Giving it a bit more time and it works.