nplan/HomeButtons #10: unavailable battery when (deep-)sleeping

Zixim, 2022-11-24:
This is a known symptom : 
[image]

The battery % is unavailable while deep-sleeping for the Home Button, yet the % is normal for my other DIY esp32 battery powered projects.

The solution is to set the birth and last will messages to empty.
References : [Esphome docs on MQTT ](https://esphome.io/components/mqtt.html#last-will-and-birth-messages) and  [Reddit thread about this](https://www.reddit.com/r/Esphome/comments/kw1g95/home_assistant_sensor_unavailable_when_esp32_is/).

EarMaster, 2022-11-24:
Do you get information from the temp and humidity sensors? My device has stopped sending these while charging the battery and while I see new values transmitted in MQTT Explorer they don't register in Home Assistant.

nplan, 2022-11-24:
Just to clarify: the device is always in deep-sleep. There is no other mode. It wakes up on a button press and every 10 minutes to post temperature, humidity and battery percentage. Then it goes back to deep-sleep.

*Last will* and *Birth* messages are not used. Home Assistant will display the sensor entities as unavailable only based on time since last received MQTT message.

This option is set in MQTT discovery sensor setup: `expire_after: '660'` Therefore the entity should never become unavailable, because messages are posted every 600 seconds normally.

@Zixim Are the button presses reliable for you? Does the battery percentage become available right after you press a button? Please check that you really receive messages every 10 minutes in MQTT Explorer or a similar app. Maybe your Wi-Fi is not reliable enough. Which Home Assistant version are you using? 

You can see two times in Home Assistant: *Last changed* and *Last updated*. But they only change if the value changes by a delta (see [here](https://www.home-assistant.io/docs/configuration/state_object/#:~:text=Note%20that%20writing%20the%20exact%20same%20state%20including%20attributes%20will%20not%20result%20in%20this%20field%20being%20updated)). Which for battery, it doesn't very often. But for expiry, the actual time of the last received MQTT message is taken into account. So the *Last changed* can be longer than 10 minutes and the entity should still be available. Given that MQTT messages are received of course.

@EarMaster Do the values you see in MQTT Explorer differ from the ones in Home Assistant?

Charging should't have any effect on how the device works and when it posts sensor messages.

EarMaster, 2022-11-24:
When I press a button the action is triggered in Home Assistant (so the connection itself works). In MQTT Explorer the values appear:
[image]

In Home Assistant the sensors register as "unavailable".
[image]

In the sensors history you can see that there the sensor received values (yesterday when I first activated the device), but the last value is too old and from around the time I plugged in the charger.
[image]

I have tried to restart the device (enter setup mode and call the /restart route, but still the values are not recognized by Home Assistant.

nplan, 2022-11-24:
@EarMaster Thanks. That's strange. I'll try reproduce it. What version of Home Assistant is this? Could you also please share what it says if you press MQTT INFO and then open payload for temperature. Should be something like this:

```
name: Home Buttons DDB400  Temperature
unique_id: HBTNS-2209-001-DDB400_temperature
state_topic: homebuttons/Home Buttons DDB400 /temperature
device_class: temperature
unit_of_measurement: °C
expire_after: '660'
device:
  identifiers:
    - HBTNS-2209-001-DDB400
platform: mqtt
```

EarMaster, 2022-11-24:
```
name: Mother Temperature
unique_id: HBTNS-2210-046-9FF628_temperature
state_topic: homebuttons/**Mother**/temperature
device_class: temperature
unit_of_measurement: °C
expire_after: '660'
device:
  identifiers:
    - HBTNS-2210-046-9FF628
platform: mqtt
```

I think I found the issue. I think I tried to change the Device Name during initial setup, but it always gets reset, when you enter setup mode again. So the sensors are listening for my new device name (_Mother_) instead of the actual device name (_Mother 9FF628_) which I at some point kept because it didn't seem worth it to change it every time.

Zixim, 2022-11-25:
> @Zixim Are the button presses reliable for you? Does the battery percentage become available right after you press a button? Please check that you really receive messages every 10 minutes in MQTT [...]

- Yesterday button presses were reliable. Today it shows "wifi error" or "mqtt error" or it works.
- When HA (**H**ome **A**ssistant) receives the button press, battery % is also updated as well as hum & temp.
- I was running MQTT explorer all day yesterday, just to monitor the button's messages, and the interval definitely wasn't 10 minutes exactly. The delay varied from 10m to 30m, which led me to believe it would only send if a certain threshold was reached. My bad.
- I'm fully certain that wifi is reliable. For now the button is in my lab/workshop, there is an AP in the same room. Also, I have dozens of other esp devices across the house - if wifi were spotty it would be pretty obvious.
- HA is version 2022.11.4 which is latest at time of writing.

nplan, 2022-11-25:
@EarMaster Exactly, that was the issue. The topic was different. 

When entering *Setup* the random ID is appended to device name each time. Thats a bug.

But when exiting the setup, MQTT discovery setup message is sent every time. So the topic in Home Assistant should update. I don't know why it hasn't in your case. If reloading the MQTT integration helped, maybe it was a problem there.

@Zixim Your issue is not related to this one. I opened a new issue and referenced your post.

closed 2022-11-26 by Zixim