nplan/HomeButtons #11: Connection not reliable

nplan, 2022-11-25:
> @Zixim Are the button presses reliable for you? Does the battery percentage become available right after you press a button? Please check that you really receive messages every 10 minutes in MQTT Explorer or a similar app. Maybe your Wi-Fi is not reliable enough. Which Home Assistant version are you using?

- Yesterday button presses were reliable. Today it shows "wifi error" or "mqtt error" or it works.
- When HA (**H**ome **A**ssistant) receives the button press, battery % is also updated as well as hum & temp.
- I was running MQTT explorer all day yesterday, just to monitor the button's messages, and the interval definitely wasn't 10 minutes exactly. The delay varied from 10m to 30m, which led me to believe it would only send if a certain threshold was reached. My bad.
- I'm fully certain that wifi is reliable. For now the button is in my lab/workshop, there is an AP in the same room. Also, I have dozens of other esp devices across the house - if wifi were spotty it would be pretty obvious.
- HA is version 2022.11.4 which is latest at time of writing.

_Originally posted by @Zixim in https://github.com/nplan/HomeButtons/issues/10#issuecomment-1327153744_

nplan, 2022-11-25:
If you get "wifi error" and "mqtt error" interchangeably, I would guess it's a Wi-Fi issue... Could you try connecting it to a different AP just to see if it works?

The interval should always be 10 minutes, but if connection fails, it will skip it. That's why the interval varies for you.

Zixim, 2022-11-26:
The AP & ESP32 are about 1.5meter apart, yet you are right about unreliable connection.
The cause is that the battery is blocking the wifi signal, well... at least at the exact angle I was positioning the Buttons on my desk.
Shifting it by just a few degrees made all the difference.

nplan, 2022-11-26:
That is interesting. I'm glad it works now.

closed 2022-11-26 by nplan