nplan/HomeButtons #18: Network connection not successful

MatthiasLohr, 2023-02-02:
Hi,

I just received my delivery with 4 home buttons. I tried to set them up, but I'm experiencing issues. I was able to set up WiFi using a smart phone, then I could connect via a browser to configure MQTT settings. Settings are correct, confirmed by a MQTT client on the computer with the browser. Home Buttons display "B1...B6" on the epaper screen.

When I now press a button, the LED behind the button lights up for ~20 seconds, then the screen flickers a lot inverting the colors several time, displays "Network connection not successful" and then back to "B1...B6" once again. Any idea?

Best regards
Matthias

Zixim, 2023-02-02:
I'd try to redo the wifi setup, folliwing this guide in the docs : https://docs.home-buttons.com/user_guide/#wifi_setup

MatthiasLohr, 2023-02-02:
I did. It says "Wifi connected :)", then restarts, then I get the IP address to access the web interface and can do the other settings, "confirming settings", "successful", "restart". The button labels are back, pressing a button: Same error after a couple of seconds.

MatthiasLohr, 2023-02-02:
Funny enough. Pressing for 10 seconds just to get back to (non-Wifi-settings) gives a WiFi error... Furthermore, screen says "software v2.0.0" but this git repository only knows v1.1.2 at the moment?

MatthiasLohr, 2023-02-02:
After downgrading to v1.1.2 it works! No idea why they're shipped with unpublished v2.0.0...

MatthiasLohr, 2023-02-02:
Unfortunately, still not working stable.

My suspicion: Can changing AP mac addresses cause a problem? What if the device is used in an environment where multiple access points exist but the one to which the device is originally connected to is not reachable anymore? Does it automatically try to reconnect to another AP? For me, it looks like that this is not happening.

nplan, 2023-02-02:
Hi Matthias,

v2.0.0 is the latest version that is shipping with the new batch of devices. The release is going live today. Shipping was apparently slightly faster than expected. 🙂 The set-up procedure has not changed though.

The device connects the first time using the SSID. If connection is successful, it stores the BSSID and channel of the AP. It uses that to connect next time to reduce the connection time. But if the connection is not successful (AP disappears or channel changes) it reverts back to using only the SSID.

What kind of Wi-Fi setup do you have? Why do you expect the APs to become unreachable frequently?

You get "Network connection not successful" if the connection (both Wi-Fi and MQTT) is not successful inside a 20s timeout. That interval is 30s in the v1.1.2 which could explain why its working on that version.

I can send you a build of v2.0.0 with a longer timeout to see if that helps.

nplan, 2023-02-02:
I created a beta release with increased timeouts for network connection. @MatthiasLohr please try if it makes any difference.

nplan, 2023-02-02:
Found the issue. Forgot to remove hardcoded credentials for my test network. 🤦‍♂️ So of course it worked for me, but not anywhere else. I'm truly sorry. Please update to v2.0.2.

@MatthiasLohr please let me know if there are still any issues due to multiple AP switching.

MatthiasLohr, 2023-02-03:
:laughing:  Set up is working fine now, and Home Assistant is discovering the button actions etc.

However, still not working is this roaming thing. I set up my devices in my office in the 2nd floor. One is to be installed in the living room (other floor) and does not see the office's AP but has its own. Even pressing several times etc., waiting, also for the regular timed updates, it does not reestablish the connection. Once I'm back in my office with the initial AP it works again like a charm.

nplan, 2023-02-03:
I recreated your case and got the same issue. Fixed the bug. Please test if it works for you.

[homebuttons_fw_v2.0.3-beta.bin.zip](https://github.com/nplan/HomeButtons/files/10582013/homebuttons_fw_v2.0.3-beta.bin.zip)

MatthiasLohr, 2023-02-05:
Thanks for the beta firmware! It look's like roaming is working now. Great work, thanks!

However, just a "feeling" (dont't have made measurements): I understand and it's totally fine that after changing the location (resulting in changing the AP) the first button press takes a longer time. But I have the (unproven, just gut feeling) impression that now follow-up button presses take a bit longer to be submitted to HA than before. But again, no idea if that is the case. Anyway: For me, the buttons are working now, thanks :)

nplan, 2023-02-05:
That's great to hear, thanks for testing!

I see no reason why subsequent presses (when the bssid and channel are already known) would take longer in the new version. The code for that hasn't changed. I also didn't detect this in my testing.

I'll close this for now.

closed 2023-02-05 by nplan