nplan/HomeButtons #22: Network issues: unreliable connection and reponse time.

mcarbonne, 2023-02-05:
When I click any button, the device fails to connect in a reasonable time.
Here is the UART output:
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x5 (DSLEEP),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6100,len:0x524
load:0x4004c000,len:0xa70
load:0x40050000,len:0x2914
entry 0x4004c18c
[   448][D][esp32-hal-tinyusb.c:673] tinyusb_enable_interface(): Interface CDC enabled
[   450][D][esp32-hal-tinyusb.c:562] tinyusb_load_enabled_interfaces(): Load Done: if_num: 2, descr_len: 75, if_mask: 0x10
[   456][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 6 = 80 Mhz, APB: 80000000 Hz
[   485][D][main.cpp:940] setup(): [DEVICE] main task started.
[   486][I][main.cpp:239] main_task(): [DEVICE] woke up.
[   486][I][main.cpp:240] main_task(): [DEVICE] SW version: v2.0.3-beta
[   491][D][state.cpp:141] load_all(): [PREF] state load all
[   515][I][main.cpp:259] main_task(): [HW] version: 2.3
[   516][I][hardware.cpp:29] init(): configured for hw version: 2.3
[   547][I][display.cpp:50] begin(): [DISP] begin
[   549][I][main.cpp:284] main_task(): [DEVICE] batt present: 1, DC connected: 0
[   550][I][main.cpp:288] main_task(): [DEVICE] batt volts: 3.747301
[   554][I][main.cpp:340] main_task(): [DEVICE] usr awake mode: 1, awake mode: 0
[   561][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=10 scl=11 freq=100000
[   570][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[   594][I][main.cpp:354] main_task(): [DEVICE] wakeup cause: PIN 5
[   595][D][buttons.cpp:16] begin(): [BTN] id 1 begun
[   595][D][buttons.cpp:16] begin(): [BTN] id 2 begun
[   599][D][buttons.cpp:16] begin(): [BTN] id 3 begun
[   603][D][buttons.cpp:16] begin(): [BTN] id 4 begun
[   608][D][buttons.cpp:16] begin(): [BTN] id 5 begun
[   613][D][buttons.cpp:16] begin(): [BTN] id 6 begun
[   618][D][main.cpp:171] start_button_task(): [DEVICE] button task started.
[   625][I][leds.cpp:9] begin(): [LEDS] begin
[   629][D][main.cpp:228] start_leds_task(): [DEVICE] leds task started.
[   635][D][main.cpp:190] start_display_task(): [DEVICE] display task started.
[   642][D][main.cpp:209] start_network_task(): [DEVICE] network task started.
[   649][D][network.cpp:30] connect(): [NET] cmd connect
[   675][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   686][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 2 - STA_START
[   688][I][network.cpp:61] update(): [NET] connecting Wi-Fi (quick mode)...
[  1124][D][buttons.cpp:103] update(): [BTN] id 1 press: SINGLE
[  1695][D][network.cpp:44] update(): [NET] state machine changed to: 2
[  1696][D][main.cpp:510] main_task(): [DEVICE] BTN_1 pressed - state SINGLE
[  1698][D][leds.cpp:30] blink(): [LEDS] blink - led: 1, blinks: 1, bri: 225, hold: 1
[  2657][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  5697][I][network.cpp:96] update(): [NET] Wi-Fi connect failed (quick mode). Retrying with normal mode... (status=0)
[  5701][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  5705][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[  6221][D][network.cpp:44] update(): [NET] state machine changed to: 3
[  6368][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  6729][I][network.cpp:111] update(): [NET] connecting Wi-Fi (normal mode): SSID: ********
[  6733][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  6735][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[  6816][D][network.cpp:44] update(): [NET] state machine changed to: 4
[  6890][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
-- 0:ttyUSB0 -- time-stamp -- févr./05/23 14:25:07 --
-- 0:ttyUSB0 -- time-stamp -- févr./05/23 14:25:22 --
[ 26811][W][network.cpp:143] update(): [NET] Wi-Fi connect failed (normal mode). Retrying...
[ 26815][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 26816][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[ 26825][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 3 - STA_STOP
[ 27346][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 27351][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 2 - STA_START
[ 27362][D][network.cpp:44] update(): [NET] state machine changed to: 3
[ 28257][I][network.cpp:111] update(): [NET] connecting Wi-Fi (normal mode): SSID: ********
[ 28259][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 28262][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 202 - AUTH_FAIL
[ 28269][D][WiFiGeneric.cpp:975] _eventCallback(): WiFi Reconnect Running
E (28184) wifi:sta is connecting, return error
[ 28291][E][WiFiSTA.cpp:278] begin(): connect failed!
[ 28301][D][network.cpp:44] update(): [NET] state machine changed to: 4
[ 30206][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
-- 0:ttyUSB0 -- time-stamp -- févr./05/23 14:25:30 --
-- 0:ttyUSB0 -- time-stamp -- févr./05/23 14:25:40 --
[ 45004][D][main.cpp:600] main_task(): [DEVICE] network connect timeout.
[ 45053][D][display.cpp:98] update(): [DISP] update: page: 5; disappearing: 1, msg: Network
connection
not
successful
[ 45114][D][buttons.cpp:30] end(): [BTN] id 1 ended
[ 45115][D][buttons.cpp:30] end(): [BTN] id 2 ended
[ 45116][D][buttons.cpp:30] end(): [BTN] id 3 ended
[ 45118][D][buttons.cpp:30] end(): [BTN] id 4 ended
[ 45122][D][buttons.cpp:30] end(): [BTN] id 5 ended
[ 45127][D][buttons.cpp:30] end(): [BTN] id 6 ended
[ 45131][D][leds.cpp:14] end(): [ 45134][I][leds.cpp:80] update(): [LEDS] ended
[: [LEDS] ended[ 45140][D][network.cpp:36] d[ 45142][I][network.cpp:120] update(): [NET] disconnecting...
[ 45151][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 45155][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[ 45164][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 3 - STA_STOP
isconnect(): [NET] cmd disconnect
[ 45172][I][network.cpp:124] update(): [NET] disconnected.
[ 45186][D][network.cpp:44] update(): [NET] state machine changed to: 1
[ 45427][D][display.cpp:56] end(): [DISP] cmd end
[ 48715][D][display.cpp:98] update(): [DISP] update: page: 1; disappearing: 0, msg: 
[ 52099][I][display.cpp:144] update(): [DISP] ended.
[ 52112][D][main.cpp:649] main_task(): [DEVICE] free stack: btns 628, disp 3620, net 8124, leds 696, main 18160, num tasks 12
[ 52113][D][main.cpp:654] main_task(): [DEVICE] free heap: esp 103100, esp min 2588, rtos 103100
[ 52121][D][state.cpp:135] save_all(): [PREF] state save all
[ 52136][I][main.cpp:51] start_esp_sleep(): [DEVICE] deep sleep... z z z
```

I did several attempts and it fails most of the time (maybe 9 out of 10, or even more).
The error pattern isn't always exactly the same but from what I have seen, "quick connect" very often fails because it did not get an IP address (but AP connection was successful). After that, falling back to "normal connect" sometimes work (but not always).

mcarbonne, 2023-02-05:
Another UART output:
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x5 (DSLEEP),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6100,len:0x524
load:0x4004c000,len:0xa70
load:0x40050000,len:0x2914
entry 0x4004c18c
[   448][D][esp32-hal-tinyusb.c:673] tinyusb_enable_interface(): Interface CDC enabled
[   450][D][esp32-hal-tinyusb.c:562] tinyusb_load_enabled_interfaces(): Load Done: if_num: 2, descr_len: 75, if_mask: 0x10
[   456][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 6 = 80 Mhz, APB: 80000000 Hz
[   485][D][main.cpp:940] setup(): [DEVICE] main task started.
[   485][I][main.cpp:239] main_task(): [DEVICE] woke up.
[   485][I][main.cpp:240] main_task(): [DEVICE] SW version: v2.0.3-beta
[   491][D][state.cpp:141] load_all(): [PREF] state load all
[   514][I][main.cpp:259] main_task(): [HW] version: 2.3
[   515][I][hardware.cpp:29] init(): configured for hw version: 2.3
[   546][I][display.cpp:50] begin(): [DISP] begin
[   548][I][main.cpp:284] main_task(): [DEVICE] batt present: 1, DC connected: 0
[   549][I][main.cpp:288] main_task(): [DEVICE] batt volts: 3.720635
[   553][I][main.cpp:340] main_task(): [DEVICE] usr awake mode: 1, awake mode: 0
[   560][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=10 scl=11 freq=100000
[   569][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[   593][I][main.cpp:354] main_task(): [DEVICE] wakeup cause: PIN 5
[   594][D][buttons.cpp:16] begin(): [BTN] id 1 begun
[   594][D][buttons.cpp:16] begin(): [BTN] id 2 begun
[   598][D][buttons.cpp:16] begin(): [BTN] id 3 begun
[   602][D][buttons.cpp:16] begin(): [BTN] id 4 begun
[   607][D][buttons.cpp:16] begin(): [BTN] id 5 begun
[   612][D][buttons.cpp:16] begin(): [BTN] id 6 begun
[   617][D][main.cpp:171] start_button_task(): [DEVICE] button task started.
[   624][I][leds.cpp:9] begin(): [LEDS] begin
[   627][D][main.cpp:228] start_leds_task(): [DEVICE] leds task started.
[   634][D][main.cpp:190] start_display_task(): [DEVICE] display task started.
[   641][D][main.cpp:209] start_network_task(): [DEVICE] network task started.
[   648][D][network.cpp:30] connect(): [NET] cmd connect
[   674][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   685][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 2 - STA_START
[   696][D][network.cpp:44] update(): [NET] state machine changed to: 3
[  1123][D][buttons.cpp:103] update(): [BTN] id 1 press: SINGLE
[  1691][I][network.cpp:111] update(): [NET] connecting Wi-Fi (normal mode): SSID: ****
[  1693][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  1696][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 202 - AUTH_FAIL
[  1703][D][WiFiGeneric.cpp:975] _eventCallback(): WiFi Reconnect Running
[  1720][D][main.cpp:510] main_task(): [DEVICE] BTN_1 pressed - state SINGLE
[  1721][D][leds.cpp:30] blink(): [LEDS] blink - led: 1, blinks: 1, bri: 225, hold: 1
[  3712][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
E (3607) esp_netif_handlers: invalid static ip
[  3731][D][network.cpp:44] update(): [NET] state machine changed to: 4
-- 0:ttyUSB0 -- time-stamp -- févr./05/23 14:34:54 --
-- 0:ttyUSB0 -- time-stamp -- févr./05/23 14:35:06 --
[ 19810][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 19811][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 19813][D][WiFiGeneric.cpp:979] _eventCallback(): WiFi AutoReconnect Running
[ 21765][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[ 23721][W][network.cpp:143] update(): [NET] Wi-Fi connect failed (normal mode). Retrying...
[ 23725][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 23726][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[ 23735][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 3 - STA_STOP
[ 24256][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 24261][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 2 - STA_START
[ 24272][D][network.cpp:44] update(): [NET] state machine changed to: 3
[ 25262][I][network.cpp:111] update(): [NET] connecting Wi-Fi (normal mode): SSID: ****
[ 25264][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 25267][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 202 - AUTH_FAIL
[ 25291][D][network.cpp:44] update(): [NET] state machine changed to: 4
[ 27218][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[... 18 lines cut ...]
[ 45370][D][buttons.cpp:30] end(): [BTN] id 4 ended
[ 45374][D][buttons.cpp:30] end(): [BTN] id 5 ended
[ 45379][D][buttons.cpp:30] end(): [BTN] id 6 ended
[ 45383][D][leds.cpp:14] end(): [LEDS] cmd end
[ 45387][D][network.cpp:36] disconnect(): [NET] cmd disconnect
[ 45402][D][display.cpp:56] end(): [DISP] cmd end
[ 45433][I][leds.cpp:80] update(): [LEDS] ended
[ 45822][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 45827][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 2 - STA_START
[ 45838][D][network.cpp:44] update(): [NET] state machine changed to: 3
[ 46829][I][network.cpp:111] update(): [NET] connecting Wi-Fi (normal mode): SSID: ****
[ 46831][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 46834][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 202 - AUTH_FAIL
[... cut ...]
[ 46881][D][network.cpp:44] update(): [NET] state machine changed to: 1
[ 50014][D][display.cpp:98] update(): [DISP] update: page: 1; disappearing: 0, msg: 
[ 53398][I][display.cpp:144] update(): [DISP] ended.
[ 53411][D][main.cpp:649] main_task(): [DEVICE] free stack: btns 628, disp 3620, net 8380, leds 696, main 18160, num tasks 12
[ 53412][D][main.cpp:654] main_task(): [DEVICE] free heap: esp 102476, esp min 2416, rtos 102476
[ 53420][D][state.cpp:135] save_all(): [PREF] state save all
[ 53432][I][main.cpp:51] start_esp_sleep(): [DEVICE] deep sleep... z z z
```

mcarbonne, 2023-02-05:
Also, I don't know if it is related or not, but if I plug my an USB-C charger, the device is rebooting in a loop:
```
[ 10810][W][main.cpp:920] main_task(): [DEVICE] free heap low, restarting...
```

In my fork, I created a branch trying to remove `String`, unwanted copies and reduce as much as possible dynamic allocation (https://github.com/mcarbonne/HomeButtons/tree/more_code_cleaning)
It does not solve the boot loop with USB cable plugged in, but my device seems to work better than before (but still not 100% reliable).

nplan, 2023-02-05:
This is definitely a memory issue. Minimal heap is far too low in your logs (esp min ~2500).

Here is my log from a working device which takes about 1 second to connect (quick mode)

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x5 (DSLEEP),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6100,len:0x524
load:0x4004c000,len:0xa70
load:0x40050000,len:0x2914
entry 0x4004c18c
[   449][D][esp32-hal-tinyusb.c:673] tinyusb_enable_interface(): Interface CDC enabled
[   461][D][esp32-hal-tinyusb.c:562] tinyusb_load_enabled_interfaces(): Load Done: if_num: 2, descr_len: 75, if_mask: 0x10
[   462][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 6 = 80 Mhz, APB: 80000000 Hz
[   491][D][main.cpp:940] setup(): [DEVICE] main ta[   491][I][main.cpp:239] main_task(): [DEVICE] woke up.
ask(): [DEV[main.cpp:240] main_tsk started.
ICE] SW version: v2.0.3
[   497][D][state.cpp:141] load_all(): [PREF] state load all
[   519][I][main.cpp:259] main_task(): [HW] version: 2.3
[   520][I][hardware.cpp:29] init(): configured for hw version: 2.3
[   551][I][display.cpp:50] begin(): [DISP] begin
[   554][I][main.cpp:284] main_task(): [DEVICE] batt present: 1, DC connected: 1
[   555][I][main.cpp:288] main_task(): [DEVICE] batt volts: 4.176508
[   558][I][main.cpp:340] main_task(): [DEVICE] usr awake mode: 0, awake mode: 0
[   565][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=10 scl=11 freq=100000
[   574][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[   598][I][main.cpp:354] main_task(): [DEVICE] wakeup cause: PIN 21
[   599][D][buttons.cpp:16] begin(): [BTN] id 1 begun
[   600][D][buttons.cpp:16] begin(): [BTN] id 2 begun
[   603][D][buttons.cpp:16] begin(): [BTN] id 3 begun
[   608][D][buttons.cpp:16] begin(): [BTN] id 4 begun
[   613][D][buttons.cpp:16] begin(): [BTN] id 5 begun
[   617][D][buttons.cpp:16] begin(): [BTN] id 6 begun
[   622][D][main.cpp:171] start_button_task(): [DEVICE] button task started.
[   629][I][leds.cpp:9] begin(): [LEDS] begin
[   633][D][main.cpp:228] start_leds_task(): [DEVICE] leds task started.
[   640][D][main.cpp:190] start_display_task(): [DEVICE] display task started.
[   646][D][main.cpp:209] start_network_task(): [DEVICE] network task started.
[   653][D][network.cpp:30] connect(): [NET] cmd connect
[   682][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   694][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 2 - STA_START
[   695][I][network.cpp:61] update(): [NET] connecting Wi-Fi (quick mode)...
[   721][D][network.cpp:44] update(): [NET] state machine changed to: 2
[   777][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  1128][D][buttons.cpp:103] update(): [BTN] id 5 press: SINGLE
[  1130][D][main.cpp:510] main_task(): [DEVICE] BTN_5 pressed - state SINGLE
[  1131][D][leds.cpp:30] blink(): [LEDS] blink - led: 5, blinks: 1, bri: 225, hold: 1
[  1448][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  1448][D][WiFiGeneric.cpp:996] _eventCallback(): STA IP: 192.168.1.60, MASK: 255.255.255.0, GW: 192.168.1.1
[  1455][I][network.cpp:80] update(): [NET] Wi-Fi connected (quick mode).
[  1461][D][state.cpp:135] save_all(): [PREF] state save all
[  1473][I][network.cpp:86] update(): [NET] SSID: planinsek, BSSID: 9C:3D:CF:2B:06:43, CH: 9
[  1474][I][network.cpp:88] update(): [NET] connecting MQTT....
[  1511][D][network.cpp:44] update(): [NET] state machine changed to: 6
[  1511][I][network.cpp:185] update(): [NET] MQTT connected.
[  1525][D][network.cpp:283] subscribe(): [NET] sub to: homebuttons/Home Buttons 265CBF/cmd/# SUCCESS.
[  1630][D][network.cpp:267] publish(): [NET] pub to: homebuttons/Home Buttons 265CBF/awake_mode/available SUCCESS.
[  1632][D][network.cpp:267] publish(): [NET] pub to: homebuttons/Home Buttons 265CBF/sensor_interval SUCCESS.
[  1641][D][network.cpp:267] publish(): [NET] pub to: homebuttons/Home Buttons 265CBF/btn_1_label SUCCESS.
[  1650][D][network.cpp:267] publish(): [NET] pub to: homebuttons/Home Buttons 265CBF/btn_2_label SUCCESS.
[  1659][D][network.cpp:267] publish(): [NET] pub to: homebuttons/Home Buttons 265CBF/btn_3_label SUCCESS.
[  1669][D][network.cpp:267] publish(): [NET] pub to: homebuttons/Home Buttons 265CBF/btn_4_label SUCCESS.
[  1680][D][network.cpp:267] publish(): [NET] pub to: homebuttons/Home Buttons 265CBF/btn_5_label SUCCESS.
[  1689][D][network.cpp:267] publish(): [NET] pub to: homebuttons/Home Buttons 265CBF/btn_6_label SUCCESS.
[  1701][D][network.cpp:267] publish(): [NET] pub to: homebuttons/Home Buttons 265CBF/awake_mode SUCCESS.
[  1705][I][WiFiClient.cpp:549] connected(): Unexpected: RES: 0, ERR: 0
[  1717][D][network.cpp:44] update(): [NET] state machine changed to: 7
[... 11 lines cut ...]
[  1830][D][buttons.cpp:30] end(): [BTN] id 6 ended
[  1835][D][leds.cpp:14] end(): [LEDS] cmd end
[  1839][  1839][I][leds.cpp:80] update(): [LEDS] ended
[  1844][I][network.cpp:215] update(): [NET] disconnecting...
[D][network.cpp:36] disconnect(): [NET] cmd disconnect
[  1859][D][network.cpp:44] update(): [NET] state machine changed to: 8
[  2873][D][network.cpp:44] update(): [NET] state machine changed to: 9
[  2877][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  2877][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[  2886][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 3 - STA_STOP
[... cut ...]
[  3037][D][state.cpp:135] save_all(): [PREF] state save all
[  3049][I][main.cpp:51] start_esp_sleep(): [DEVICE] deep sleep... z z z
```

Notice the min free heap is **62576**. It's also not any lower if quick connect fails and it retries using normal mode.

I did quite some testing to ensure that there is always enough free heap. Wi-Fi stack starts to fail at around 10k free and lower.

Did you make any other changes to the code that could cause this? What happens if you flash the latest bin firmware from releases that I built?

mcarbonne, 2023-02-05:
I just flashed the official "2.0.3" binary and got the same bootloop issue with USB-C cable plugged-in:
```
[ 10734][W][main.cpp:921] main_task(): [DEVICE] free heap low, restarting...
```
and buttons are still not responsive at all while running on battery (network connection issues).

In my custom branch, I registered a callback to trace failed allocations (`heap_caps_register_failed_alloc_callback(heap_alloc_failed_callback)`) and there are a lot of them.

Maybe it is related to my WiFi network requiring more resources ? Stronger encryption ? It was setup as WPA2 PSK/WPA3-SAE, and I also tried WPA2-PSK only.
What is the release process to create "official" builds ? For my tests, I'm using PlateformIO in vscode to build, and then I retrieve the binary found here: `.pio/build/homebuttons/firmware.bin`

nplan, 2023-02-05:
With the USB-C plugged in the device does not go to sleep. I programmed it to restart if min free heap is lower than 20k since there is a memory leak in the Wi-Fi (or maybe MQTT) library that I couldn't debug yet. But it should happen only once every 12 hours not right away. 

There's no special build process, I do it exactly like you. I just wanted to rule it out. 

Would be great if we could somehow figure out which part of code is allocating so much memory. In your case it's 60KB more than in mine which is a lot.

Can you please try to flash it again but first fully erase the flash with esptool. Maybe there is a problem with loading preferences from flash.

mcarbonne, 2023-02-05:
Concerning preferences in flash, I added some logs to check what was loaded. Everything seems fine.

Also, when using the official 2.0.3 release, it was working better than when I was building 2.0.3 by myself (but still not perfect).
So maybe your dependency versions aren't the same as mines.
I edited `platformio.ini` like this to force precise versions :
```diff
-       knolleary/PubSubClient@^2.8
-       bblanchon/ArduinoJson@^6.19.4
+       knolleary/PubSubClient@2.8
+       bblanchon/ArduinoJson@6.19.4
        https://github.com/tzapu/WiFiManager.git#v2.0.13-beta
-       adafruit/Adafruit SHTC3 Library@^1.0.0
-       zinggjm/GxEPD2@^1.4.7
-       ricmoo/QRCode@^0.0.1
-       olikraus/U8g2_for_Adafruit_GFX@^1.8.0
+       adafruit/Adafruit SHTC3 Library@1.0.0
+       zinggjm/GxEPD2@1.4.7
+       ricmoo/QRCode@0.0.1
+       olikraus/U8g2_for_Adafruit_GFX@1.8.0
        https://github.com/Neargye/semver.git
-       https://github.com/adafruit/Adafruit_SSD1327.git
-       adafruit/Adafruit SSD1327@^1.0.4
+       adafruit/Adafruit SSD1327@1.0.4
```
For now, it seems to work way better. I'm running 40323805ab3a93b6e3a1acb32c46397384b76943 (https://github.com/mcarbonne/HomeButtons/tree/more_code_cleaning) and it is almost usable (working 90% of the time in ~1s).
My branch also contains optimizations to reduce memory use.

But nevertheless, all versions (official builds and my customs builds) are facing the same issue when usb-C is plugged.

nplan, 2023-02-06:
Optimising other parts of code won't fix the issue. There is a memory leak somewhere that is causing the Wi-Fi to have problems. We should focus on that.

I want to replicate this on one of my devices. Could you send me a flash dump of your device with the build that is performing the worst for you?

This is the command to dump it: `python -m esptool -b 921600 read_flash 0x400000 flash_dump.bin`

Here is a build log with dependencies in my environment:
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/homebuttons_rev1.0.html
PLATFORM: Espressif 32 (6.0.0) > HomeButtons Rev 1.0
HARDWARE: ESP32S2 80MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6) 
 - tool-esptoolpy @ 1.40400.0 (4.4.0) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 46 compatible libraries
Scanning dependencies...
Dependency Graph
|-- PubSubClient @ 2.8.0
|-- ArduinoJson @ 6.20.0
|-- WiFiManager @ 2.0.13-beta+sha.1545992
|   |-- DNSServer @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- ESPmDNS @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- Update @ 2.0.0
|   |-- WebServer @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- Adafruit SHTC3 Library @ 1.0.1
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- Adafruit Unified Sensor @ 1.1.7
|-- GxEPD2 @ 1.5.0
|   |-- Adafruit GFX Library @ 1.11.5
|   |   |-- Adafruit BusIO @ 1.14.1
|   |   |   |-- Wire @ 2.0.0
|   |   |   |-- SPI @ 2.0.0
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|-- QRCode @ 0.0.1
|-- U8g2_for_Adafruit_GFX @ 1.8.0
|   |-- Adafruit GFX Library @ 1.11.5
|   |   |-- Adafruit BusIO @ 1.14.1
|   |   |   |-- Wire @ 2.0.0
|   |   |   |-- SPI @ 2.0.0
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|-- semver @ 0.0.0+20230127210511.sha.eae828a
|-- Adafruit SSD1327 @ 1.0.4+sha.6404a44
|   |-- Adafruit GFX Library @ 1.11.5
|   |   |-- Adafruit BusIO @ 1.14.1
|   |   |   |-- Wire @ 2.0.0
|   |   |   |-- SPI @ 2.0.0
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|-- Adafruit SSD1327 @ 1.0.4
|   |-- Adafruit GFX Library @ 1.11.5
|   |   |-- Adafruit BusIO @ 1.14.1
|   |   |   |-- Wire @ 2.0.0
|   |   |   |-- SPI @ 2.0.0
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|-- Adafruit GFX Library @ 1.11.5
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|-- Wire @ 2.0.0
|-- WiFi @ 2.0.0
|-- Preferences @ 2.0.0
Building in release mode
Retrieving maximum program size .pio/build/homebuttons/firmware.elf
Checking size .pio/build/homebuttons/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  15.8% (used 51620 bytes from 327680 bytes)
Flash: [========  ]  77.3% (used 1012978 bytes from 1310720 bytes)
```

nplan, 2023-02-06:
Also tried it today on a WPA3-AES network (TP-Link Archer C54). It does require some more memory (59k min free heap vs. 62k on WPA2-PSK) but otherwise works as expected. The connection time is about a second longer though.

mcarbonne, 2023-02-07:
After a lot of digging, I found out the root cause of my memory leak: ARP !
On my network, a periodic ARP scan is running in background (something like `nmap -F -T4 --min-rate 10 --host-timeout 5s XXX.XXX.XXX.XXX/24`). It seems that the amount of generated ARP (broadcasted by definition) is causing trouble. I don't know if the trouble is caused by the ARP requests or the process of gratuitous ARP in answer to requests.
When disabling this background scan:
- awake mode is working (no more boot loop)
- "normal" mode is working way better too

Also, is it possible to debug over USB ? (as my JTAG probe is broken) ? It would have spare me a lot of time debugging (ie. finding that the network stack was the culprit).

But my work on reducing memory footprint of the software also shown significant improvements (it was almost working in awake mode while being spammed by ARP requests). I'll submit a PR after cleaning what I've done.
Also tweaking the WiFi stack is an option: on my network, there are other devices based on esphome (esp8266) which are running fine with all those ARP packets.
There is some official documentation about this topic:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/wifi.html#using-psram
I'll have a look when I have time.

Zixim, 2023-02-08:
@mcarbonne Out of personal curiosity, would you mind elaborating a little on the reasons behind this ARP scan ?
I'm absolutely not being critical, just wondering if there's something in there that I could learn from...

mcarbonne, 2023-02-08:
> @mcarbonne Out of personal curiosity, would you mind elaborating a little on the reasons behind this ARP scan ? I'm absolutely not being critical, just wondering if there's something in there that I could learn from...

I'm using https://www.home-assistant.io/integrations/nmap_tracker/ in my home assistant setup.
So far, it is the most reliable way to identify connected devices and trigger automations.

nplan, 2023-02-12:
@mcarbonne I can't replicate this on my network. I added the Nmap Tracker integration in Home Assistant with the default settings. The behaviour of Home Buttons is still normal. Using Wireshark, I see around 20 ARP packets per second being published from the Home Assistant server.

Are you using any specific nmap parameters?

mcarbonne, 2023-02-13:
The scan performed by nmap_tracker integration is similar to something like this
```bash
while [ true ]
do
   nmap -F -T4 --min-rate 10 --host-timeout 5s 192.168.0.0/24
   sleep 3
done
```
(default options are available here: https://github.com/emilburzo/core/blob/dev/homeassistant/components/nmap_tracker/const.py)
Note: this will perform a scan every ~10 seconds because nmap take ~7 seconds to scan the network.

Also it seems that there is more than one problem.
Continuous network scan is for sure:
- triggering the boot loop when power is plugged
- triggering most of (or even maybe all) extra memory allocations causing ESP running out of RAM.

But sometimes, I also encounter issues when this scan is disabled. The ESP is connecting well to my access point but never obtains an IP address. I don't have enough evidence yet to open a dedicated issue. It might be related to either my access point or the ESP dhcp client.

nplan, 2023-02-15:
Did some more testing today using ping.

I can trigger reboot due to low min free heap (<20K) consistently with this command:

`sudo ping -c 26 -i 0.001 192.168.2.2` = 26 packets sent sequentially with 1 ms in between

It seems that the problem is a large amount of network packets received in a short time. The size of the ping request packet does not have an impact.

nplan, 2023-02-22:
The issue with low min free heap is due to the ESP32 Wi-Fi dynamic buffers being too large. They can take up to 70k memory if receiving large number of packets in a short time. 

```
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=12
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=12
```

The fix is to reduce this config parameters from default of 32 to 12. Now the worst case dynamic memory allocation by Wi-Fi driver is around 15k.

Getting access to this configuration options required to switch the PlatformIO framework from `arduino` to `espidf` with `arduino` as a component.

@mcarbonne I just pushed this changes to dev (commit f077866) if you want to try it.

mcarbonne, 2023-02-24:
@nplan I confirm that your latest commit on dev fix my first issue. No more boot loop in awake mode !
I tested directly in my work (commit c2691e1324768f4d4cb2d74d44f05ecbaf0e2eb0).

closed 2023-02-24 by mcarbonne