nplan/HomeButtons #23: Buttons labels not appearing in HA

frediz, 2023-02-07:
Hi and thanks for the great piece of hardware and software!
I installed the latest 2.0.3 firmware and the device seems to work well.
I got temperature, humidity and some other reported to the mqtt server and available in HA,
but the button controls don't appear. "Awake mode" and "Sensor interval" controls do appear.
I tried changing the labels in setup mode hoping it will trigger something somewhere but it didn't. Pressing buttons neither.
I didn't have time to monitor the mqtt communications so far and maybe I'm missing some trivial point.
Thanks in advance,
F.

nplan, 2023-02-07:
What version of Home Assistant are you running? Text input for labels is only supported on 2022.12 or later.

frediz, 2023-02-08:
Hi Nejc,
thanks for the tip, I was on 2022.11.x . After upgrading it worked right away!
F.

closed 2023-02-08 by nplan