nplan/HomeButtons #27: Blueprint for easier HA integration

MatthiasLohr, 2023-02-24:
Hi there,

is somebody aware of a blueprint for an easier set-up with HA? Would be great to have something similar to https://epmatt.github.io/awesome-ha-blueprints/docs/blueprints/controllers, so I don't have to create 18 automations manually (6 buttons á 1, 2 and 3 presses).

Best regards
Matthias

NicoDreamzZ, 2023-03-21:
I tried building a blueprint, but it seems kinda complicated. The firmware uses mqtt discovery. This creates mqtt device triggers at home assisted. I found no functional way to implement that in a blueprint.

As a simple solution for not having multiple automations, I wrote all triggers into one automation, gave them each and checked later in the conditions.
[image]

If you need it, i can share my yaml code.

MatthiasLohr, 2023-03-21:
> If you need it, i can share my yaml code.

That would be great. Maybe I can start from there.

NicoDreamzZ, 2023-03-21:
I gave the triggers all a unique id, and in the actions I use the if and check which trigger was used.
```
alias: "[Action] [Button] HomeButtons"
description: ""
trigger:
  - platform: device
    domain: mqtt
    device_id: f1c7da1df4e124d045f9b76b9ddf9adf
    type: button_short_press
    subtype: button_1
    discovery_id: HBTNS-2301-180-31FD62 button_1
    id: 1_single
  - platform: device
    domain: mqtt
    device_id: f1c7da1df4e124d045f9b76b9ddf9adf
    type: button_short_press
    subtype: button_2
    discovery_id: HBTNS-2301-180-31FD62 button_2
    id: 2_single
  - platform: device
    domain: mqtt
    device_id: f1c7da1df4e124d045f9b76b9ddf9adf
    type: button_short_press
    subtype: button_3
    discovery_id: HBTNS-2301-180-31FD62 button_3
    id: 3_single
  - platform: device
    domain: mqtt
    device_id: f1c7da1df4e124d045f9b76b9ddf9adf
    type: button_short_press
    subtype: button_4
    discovery_id: HBTNS-2301-180-31FD62 button_4
    id: 4_single
  - platform: device
    domain: mqtt
    device_id: f1c7da1df4e124d045f9b76b9ddf9adf
    type: button_short_press
    subtype: button_5
    discovery_id: HBTNS-2301-180-31FD62 button_5
    id: 5_single
  - platform: device
    domain: mqtt
    device_id: f1c7da1df4e124d045f9b76b9ddf9adf
    type: button_short_press
    subtype: button_6
    discovery_id: HBTNS-2301-180-31FD62 button_6
    id: 6_single
  - platform: device
    domain: mqtt
    device_id: f1c7da1df4e124d045f9b76b9ddf9adf
    type: button_double_press
    subtype: button_3
    discovery_id: HBTNS-2301-180-31FD62 button_3_double
    id: 3_double
condition: []
action:
  - if:
      - condition: trigger
        id: 1_single
    then:
      - service: input_boolean.toggle
        data: {}
        target:
          entity_id: input_boolean.gaste
  - if:
      - condition: trigger
        id: 2_single
    then:
      - event: test.test.test.test
        event_data: {}
  - if:
      - condition: trigger
        id: 3_single
    then:
      - service: cover.open_cover
        data: {}
        target:
          entity_id:
            - cover.405297
            - cover.40528b
  - if:
      - condition: trigger
        id: 3_double
    then:
      - service: cover.close_cover
        data: {}
        target:
          entity_id:
            - cover.405297
            - cover.40528b
  - if:
      - condition: trigger
        id: 4_single
    then:
      - service: script.fix_thermostate
        data: {}
      - delay:
          hours: 0
          minutes: 0
          seconds: 3
          milliseconds: 0
      - service: script.thermostate_eco
        data: {}
  - if:
      - condition: trigger
        id: 5_single
    then:
      - service: script.fix_thermostate
        data: {}
      - delay:
          hours: 0
          minutes: 0
          seconds: 3
          milliseconds: 0
      - service: script.thermostate_komfort
        data: {}
  - if:
      - condition: trigger
        id: 6_single
    then:
      - service: script.thermostate_aus
        data: {}
mode: single
```
Here some screenshots:
[image]
[image]

Curtis-777, 2024-11-09:
> I gave the triggers all a unique id, and in the actions I use the if and check which trigger was used.
>
> ```
> alias: "[Action] [Button] HomeButtons"
> description: ""
> trigger:
>   - platform: device [...]

Thank you, can you explain how to use this ? 
I cant get IT running right

NicoDreamzZ, 2024-11-13:
> Thank you, can you explain how to use this ? I cant get IT running right

At the moment i don't use my HomeButtons. I'm not sure if my code still works. You have to edit the entity ids to your specific installation.