nplan/HomeButtons #29: [Feature] Add support for icons

N1c093, 2023-02-27:
Hello,

it would be great, when the display also support mdi icons. 
This would save some space for letters on the display.

Thank you

nplan, 2023-03-13:
I'm planning to add icons soon. But I can only fit ~1400 icons on the device, not the whole mdi set.

I was thinking of adding the 646 icons in the `Home Automation` [category](https://pictogrammers.com/library/mdi/category/home-automation/) at first. Do you think that's sufficient or should I add another category?

hilsonp, 2023-03-14:
This is a really good start in any case. 
I thought the small arrow category would be useful too. 

https://pictogrammers.com/library/mdi/category/arrow/

N1c093, 2023-03-14:
@nplan this sounds like a good start 👍

I would also agree with @hilsonp that some arrows would be helpful.

mcarbonne, 2023-03-14:
> I'm planning to add icons soon. But I can only fit ~1400 icons on the device, not the whole mdi set.
>
> I was thinking of adding the 646 icons in the `Home Automation` [...]

Rather that storing icons on the device, wouldn't it be possible to send icons on the fly to the device when they are in use, over mqtt maybe ? I'm not very familiar with home assistant API but I assume there is a way to replace the label field by an "image" field or something similar.

nplan, 2023-03-15:
@mcarbonne it could be done, but would require a custom integration in HA. Which would add additional setup steps. Also, the users who use device without HA (although I doubt there's many) would not be able to use the feature.

nplan, 2023-04-25:
MDI added in v2.1.0.

closed 2023-04-25 by nplan