nplan/HomeButtons #30: Example for Homebridge?

jiserra, 2023-03-06:
Hi! I bought Home Buttons and even though I have Homebridge instead of Home Assistant, it has MQTT plugins. My idea was to setup a switch in Homebridge and control it with Homebuttons, but I can't seem to make it work. I think my lack of understanding of how MQTT works is part of the problem, but also, I don't know how to make auto discovery work on Homebridge. I appreciate any pointers on trying to solve this, thanks!

nplan, 2023-03-06:
Hi,
Autodiscovery is specific to Home Assistant so it won't work with other platforms. You need to set it up manually. Looking briefly into Homebridge the steps to set it up would be:

1. Install an MQTT broker (I suggest https://mosquitto.org/. You could install it on the same server that runs Homebridge)
2. Configure Home Buttons to use the MQTT broker using the setup guide
3. Verify that it works using http://mqtt-explorer.com/. Home Buttons uses these topics: https://docs.home-buttons.com/mqtt_topics/
5. Install Homebridge MQTT plugin (https://www.npmjs.com/package/homebridge-mqttthing this one seems fine)
6. Configure the plugin with this device type: https://github.com/arachnetech/homebridge-mqttthing/blob/HEAD/docs/Accessories.md#statelessprogrammableswitch. Set the topics to the ones you see in MQTT-explorer.

I think it should work. Let me know how it goes.

jiserra, 2023-03-08:
Wow, that was such a great answer!! It all worked out! I had to fiddle a little bit with the configuration of the plugin in Homebridge but nothing that reading documentation wouldn't answer. This is great, thank you so much!

closed 2023-03-08 by jiserra