nplan/HomeButtons #35: Unreliable connection and reponse time with FW 2.0.4 and WPA2-PSK/WPA3-SAE WiFi

PaulRitzkat, 2023-03-14:
Hi,

I've flashed the new 2.0.4 firmware, but there is still a delay of around 10 to 18 seconds before Homeassistant triggers/the MQTT connection is established and the button light goes off after I press a button. My setup consists of a TP-Link EAP610 located 2 meters away with direct sight and is secured with WPA2-PSK/WPA3-SAE.

When I use a temporary WiFi network from my Fritbox, which uses WPA2+WPA3 for security by default, it takes around 3 to 4 seconds to connect and send the MQTT message. If I drop the security to WPA2 (CCMP), it takes around 2 seconds.

I am willing to provide additional logs via UART, but I would need some guidance on how to do so.

nplan, 2023-03-15:
I guess it's taking that long to obtain the IP from you router... What router setup do you have? EAP610 is just an access point if I'm correct?

Would be really helpful if you could provide logs. There's a debug "connector" next to reset button. You can connect a USB to UART adapter to the TX and RX pins and then use a serial monitor. The baud is 115200. But please flash (with OTA) the attached  debug build first so the log will be more verbose.

[homebuttons_fw_v2.0.4_debug.bin.zip](https://github.com/nplan/HomeButtons/files/10977045/homebuttons_fw_v2.0.4_debug.bin.zip)
(extract the zip first)

PaulRitzkat0110, 2023-03-15:
The EAP610 is just an AP, thats correct. The fritzbox is the DHCP server and is set to always give the same IP to the homebutton esp. Can I set a static IP? With WPA+WPA2 the connection time is under 2 seconds, I would says ~1,5

I flashed the debug firmware, but I don't get any output on serial (with screen /dev/ttyUSB0 115200, even tried switching RX/TX pins). I checked with an ESP8266 and that one worked with my FTD1232.
[image]

nplan, 2023-03-15:
Sorry, gave you wrong instructions.... You should connect TX pin of Home Buttons to RX pin of UART adapter and GND to GND. 

I think you should also move the jumper on you adapter. It should be set to 3.3v.

Please try again.

PaulRitzkat, 2023-03-15:
GND and TX/RX worked flawlessly :) I attached each a reset and button press from my fritzbox IOT WiFi with WPA+WPA2 and one from my EAP Skynet WiFi with WPA2+WPA3 (two button presses for good measure)
[homebutton_iot.log](https://github.com/nplan/HomeButtons/files/10983833/homebutton_iot.log)
[homebutton_skynet.log](https://github.com/nplan/HomeButtons/files/10983834/homebutton_skynet.log)

NicoDreamzZ, 2023-03-15:
Would it be an advantage if we provided a static IP configuration?

PaulRitzkat, 2023-03-15:
Thank you for your time! I don't have any "security" settings regarding reconnect times or similar, but I've got logs in the omada controller regarding the ESP connects:

> Mar 15, 2023 23:21:59    esp32s2-A44230 is disconnected from SSID "Skynet" on AP Flur (1m connected, 2.70KB).
> Mar 15, 2023 23:15:58    [Failed]esp32s2-A44230 failed to connected to AP Flur with SSID "Skynet" on channel 11 because WPA Authentication times out/failed.(1 time in a minute)
> Mar 15, 2023 23:15:08    esp32s2-A44230 is connected to AP Flur with SSID "Skynet" on channel 11.

Logfile with a couple button presses with >60s wait (second/third time didn't take 20s, more ~8s): [homebutton.log](https://github.com/nplan/HomeButtons/files/10984966/homebutton.log)

Logfile with failure, auto wakup from sleep to send data: [homebutton_sleep.log](https://github.com/nplan/HomeButtons/files/10985075/homebutton_sleep.log)

nplan, 2023-03-16:
The first log looks fine. At first press it does the normal slow connect because it was reset prior to that. On the next two presses it uses quick connect (with bssid and channel) which succeed normally. It does take a bit longer than it should because of two reasons: 1. this is debug build which boots slower, 2. due to WPA3 (I'm looking into if this could be improved)

In the second log the quick connect fails due to `AUTH_EXPIRE` which I'm not sure what it means. But then it connects using normal slow mode as expected.

Did you try rebooting your router and access point?

Also I randomly found this: https://community.home-assistant.io/t/esphome-wifi-auth-expired/443790/28
The poster has basically the same situation: using channel 11 and is getting both `AUTH_EXPIRE` and `4WAY_HANDSHAKE_TIMEOUT` exactly like in your case.

Please try changing the channel to see what happens.

hilsonp, 2023-03-16:
Sorry if you see this as a hijack but I have been looking for static up configuration also and ended up surprise that HB does dhcp “only”. Do not get me wrong. dhcp is the easiest for most people and makes your device foolproof. Nevertheless, dhcp has a cost (which is maybe small compared to wifi négociation, I have no idea) and for a device that you would want to send asleep asap, I think allowing static ip configuration would be good. My 2 cents 😉

nplan, 2023-03-16:
@hilsonp Sure, I'll add static IP in the future.

PaulRitzkat, 2023-03-16:
Interesting with CH11, I rebootet the AP and switched to CH6 (20MHz width), but sadly no luck with quicker logins.

all slow connects: [homebutton.log](https://github.com/nplan/HomeButtons/files/10995693/homebutton.log): reset; button press; ~60s wait; button press (~20s); ~80s wait; button press (~30s); 20s wait; button press (~34s); wake up from sleep
and one log from a quick connect: [homebutton_quick.log](https://github.com/nplan/HomeButtons/files/10995713/homebutton_quick.log)

Same errors in the AP logs: [Failed]esp32s2-A44230 failed to connected to AP Flur with SSID "Skynet" on channel 6 because WPA Authentication times out/failed.(4 times in a minute)

Most of the time its slow (>10s) and sometimes its quicker and immediately after its another slow connect. Channel utilization is between 30% and 40% and the area here has lots of other 2.4GHz networks. If I move the homebutton directly under the AP the chance of quicker connect gets better.

Can I solder an antenna somewhere to see if that helps? I can run a separate IOT WiFi directly from my fritzbox and can use it, but fast WPA3 connects would be nice :)

nplan, 2023-03-18:
Are those two different networks in the last two logs or did you edit the ssid and bssid in one of them to hide it?

There's nowhere to connect an external antenna unfortunately.

If you can use another network that's the best option right now. I'm looking into it but I don't have a solution yet.

PaulRitzkat, 2023-03-18:
Oh, yeah those are the same networks, I forgot to hide it :D 

No worries, I'll use a dedicated IOT network in the meantime. If you need someone to test something, just ping me :)

nplan, 2023-03-20:
Please try this build. I'm not sure if it will solve your issue, but it reduces the connection time by about 40% for both WPA2 and WPA3. 

In testing I get the average connection times:
- WPA2: **980** ms
- WPA3-SAE: **2665** ms

[homebuttons_fw_v2.0.5-beta.bin.zip](https://github.com/nplan/HomeButtons/files/11022368/homebuttons_fw_v2.0.5-beta.bin.zip)
(extract the zip first)

Zixim, 2023-03-21:
I also just tried it, and there's indeed a serious difference in the connection speed. Great job !!
Next thing that will make a real difference is setting a fixed IP in the device. (which is a feat. that was already req'd in another post)

PaulRitzkat, 2023-03-23:
Thats better :) I still sometimes get longer delays, but that might be the wall the box is mounted to (or other 2.4GHz networks interfering).

nplan, 2023-03-25:
I released v2.0.5 with this fix and also added support for setting static IP, which should further reduced the reaction time.

closed 2023-03-25 by nplan