nplan/HomeButtons #37: Button Lable change Iobroker via MQTT

Frodonet, 2023-03-20:
Hi I tried your great Produkt. Alls works fine with my IObroker but to change the Button Labels I didi not find a working way. 
Can you help me to get a working url Request ?

Regards Frodonet

Taffmailer, 2023-03-20:
Hi
I would like to join this - **super project**.
I have the same problem and looking for a solution how I can change the label via mqtt in the iobroker environment.

all the best

Zixim, 2023-03-20:
Home Assistant sends a MQTT message to `homebuttons/Home Buttons XXXXXX/cmd/btn_6_label` with payload containing the label.
XXXXXX being the device identifier.

Frodonet, 2023-03-20:
ok I can send the Message to the device buy instead of my text it shows "null"  on the display

nplan, 2023-03-20:
`{BASE_TOPIC}/{DEVICE_NAME}/cmd/btn_{1-6}_label` is the topic used for label text. The default `BASE_TOPIC` is `homebuttons` and `DEVICE_NAME` is what you set it during initial setup. Defaulting to `Home Buttons XXXXXX` with Xs replaced with the ID. The message should be published with retained flag set, so that Home Buttons receives it when it wakes up.

Since you're getting "null", could you please check with [MQTT Explorer](http://mqtt-explorer.com/) what is actually published to the topic?

Taffmailer, 2023-03-22:
I have published a value (see the screenshot in the attachment). After the „home button" was woken up by pressing any button, the data communication took place, which I have screenshot logged. (Sorry for this format, I have not found a full logfile about the history).
I hope that this is helpful

[image]
[image]

Frodonet, 2023-03-22:
I have the same in my MQTT Explorer.

Taffmailer, 2023-03-22:
I'm using iobroker with the adapter MQTT 4.0.7

Taffmailer, 2023-03-22:
Thanks a lot, I've solved the problem.

- installed mosquitto at iobroker server, 
- created a separate mqtt instance at iobroker in client mode which connects to mosquitto
- Topic is homebutton/# 
- configured the retainer flag there
- and reconfigured the home button (new mqtt broker is now mosquitto)

... and it works

closed 2023-03-22 by nplan

Frodonet, 2023-03-23:
> Thanks a lot, I've solved the problem.
>
> * installed mosquitto at iobroker server,
> * created a separate mqtt instance at iobroker in client mode which connects to mosquitto
> * Topic is homebutton/#
> * [...]

Can you tell me more details ??