nplan/HomeButtons #38: Problem when i first powered the Device with USB C

marthaler, 2023-03-20:
The Battery is not here already, so i powered the Device by USB-C. All went ok, was able to integrate with Home Assistant. The labels i was able to change. All ok so far. When i removed USB-C and repluged, Display showed Reboot, the Display flashed Black / white. And the Labels hav the Values, thadvi changet to. Problem now is, no Light, when i Push a Button, not changing to Temperature, when i Push for 30 Sec, no Reset is happening. Is this a known Bug. Is there a Solution for my Problem? Best Regards Daniel

nplan, 2023-03-20:
Could you please answer this questions so I can better understand the issue:
1. Does it display `RESTART... ` if you unplug and replug USB-C?
2. Does it display `RESTART... ` if you press RST button inside the case on the PCB?
3. Is the display refreshing at all in any case?
4. Did you try using a different power brick or connecting it to a computer USB port?

marthaler, 2023-03-21:
Thank you for your Help!

> 1. Does it display `RESTART... ` if you unplug and replug USB-C?

-  Yes

> 2. Does it display `RESTART... ` if you press RST button inside the case on the PCB?
-  Yes
> 3. Is the display refreshing at all in any case?
- Yes, after `RESTART... ` it refreshes several Times (turning from inverted / black & white ) for maybe 5 Times.
- - It is not sending any Messages to MQTT and i cant' find it in my Network¨
> 4. Did you try using a different power brick or connecting it to a computer USB port?

hugosbnarciso, 2023-03-21:
i couldn’t even reach the setup part when it was only connected via usb-c, once i plugged the battery, it worked as intended. Maybe try that

nplan, 2023-03-21:
@marthaler when you press any of the 6 buttons nothing happens? LED does not blink and the display doesn't refresh?

Could you please flash this to the device using esptool:
[2301-174_v2.0.4.bin.zip](https://github.com/nplan/HomeButtons/files/11031413/2301-174_v2.0.4.bin.zip)
(extract the zip first)

The command is: `python -m esptool --port PORT --after no_reset write_flash 0x 02301-174_v2.0.4.bin`

Put the device in boot mode by pressing the reset button while holding the boot button. LED under lower left button will light up. Replace `PORT` in the command with the serial port on your computer or omit `--port` to detect it automatically.

You will have to do the setup again after flashing.

marthaler, 2023-03-21:
@nplan: Yes that was the case, no LED. I was was just beginning the flashing with the Version you provided. And i was just pluging the device in one more Time. With another USB Cable directly to my Computer... And now everything is working as expected.
So i will not flash it. Unless it is useful for you...?
One more thing: I tested in different WLAN's with the Same ID / and Password. Maybe my Problem was because of that. I will for sure test again in all of this Networks.

Thank you so much: Great work, great Product. I will order some more if available Again :-)

marthaler, 2023-03-21:
Thank you for Help & Comment