nplan/HomeButtons #42: [feature] Set button labels via MQTT

anon8675309, 2023-04-06:
As a user, I would like to be able to set the button labels without touching the device.

Here's an example of a command I would expect to cause the label for button 6 to be updated within N minutes:

```sh
mosquitto_pub -h 192.168.0.109 -p 1883 -t "homebuttons/Home buttons/btn_6_label" -m "PARTY"
```

I saw that the device can [subscribe](https://github.com/anon8675309/HomeButtons/blob/master/Firmware/HomeButtonsArduino/src/main.cpp#L151) to events, so I would expect that this should be possible even when operating in sleep mode.

Note: I am not currently using HomeAssistant and am not planning on doing so. I am interacting directly with the HomeButton over MQTT.

nplan, 2023-04-06:
This is already supported. The device wakes up every N minutes (configurable) to publish sensor states and check for new commands.

You are missing a retain flag in your publish command. Try this:

`mosquitto_pub -h 192.168.0.109 -p 1883 -t "homebuttons/Home buttons/cmd/btn_6_label" -m "PARTY" -r`

The device cannot receive messages when sleeping. The retain flag makes the MQTT broker store the message and send it to device when it reconnects.

anon8675309, 2023-04-07:
This worked perfectly, thanks!  I wrote up some documentation about testing directly with MQTT and will submit a pull request so it can be added to the official repo if it looks acceptable.

closed 2023-04-07 by anon8675309