nplan/HomeButtons #44: Availability of HomeButtons to buy

gon0, 2023-04-16:
Dear @nplan, 
you convinced me with your great product and I would like to buy more 😁 
The availability in Tindie is only one pcs. The link to Lectrons (https://docs.home-buttons.com/#where-to-get) is not actual, but on lectronz, the availability is zero.

For me it is not urgent. Will there be more devices available soon? It would be ok for me to preorder, so you can plan ahead.

Keep up your great work! :)

nplan, 2023-04-16:
I'm glad to hear that. 🙂 Thanks for your interest.

More will be available in about a month. Preorders are unfortunately not possible at the moment.

gon0, 2023-04-16:
Very well! I am glad to hear you are keeping up your good work and support! 

I'll check back in a month :)

closed 2023-04-16 by gon0