nplan/HomeButtons #46: MDI icons do not display

daj102, 2023-04-25:
Was very excited when I saw the icon feature, so I've just tried it out ... unfortunately, they do not display.  They only show a document/question mark icon instead.

[image]

I tried it with the following labels:

* `mdi:play-pause Music`
* `mdi:play-pause`
* `mdi:bedroom` (from the [documentation](https://docs.home-buttons.com/user_guide/#icons), so I thought the icon would work)
* `mdi:bedroom Music`

I clicked one of the buttons and the screen briefly said 'Downloading icons ...' but nothing changed.  The device does have internet access.

Is there anything I can do to debug it?

nplan, 2023-04-25:
`mdi:bedroom` is **not** a MDI icon so this is expected behaviour. It's also not mentioned anywhere in the documentation.

But the `mdi:play-pause` it should work. Have you tried any other icons beside these two?

Did you get `Icon server NOT reachable` error after `Downloading icons...` or not?

Have you tried restarting the device using the settings [menu](https://docs.home-buttons.com/user_guide/#settings)?

nplan, 2023-04-25:
Are you using _Home Assistant_ to set labels or the web interface?

daj102, 2023-04-25:
I've tried both ...

So oddly, as I was writing this, the device just refreshed and now the icons are showing up. I had it in setup mode, but then got on a call for work and I guess it timed out - and when the screen refreshed, the icons were there.  Lemme do some more testing.

daj102, 2023-04-25:
so I just added `mdi:sun-clock-outline` through the setup website, and it shows as `?` after the save process.  I went into the new menu (very nice, btw) and restarted it - still `?`.  I'll put it into setup mode and see if it times out ...

daj102, 2023-04-25:
It did time out, but the icon was still `?`.  Only then did I press a button and got `Downloading icons ...` and now it is correct.  I think this is just a timing thing ... maybe you could force it to validate the icons and download them right after the setup finishes, in case the labels have changed?  And as you mentioned HomeAssistant, I tried changing the icons there again and the labels didn't change until I pressed a button and it connected back to MQTT.

Do you want me to close this issue, or would you prefer to keep it open and look into the process timing?

nplan, 2023-04-25:
This is the expected behaviour. When setting icon in web setup and closing the setup the icons are not downloaded yet (therefore the `?`). The download happens on the next button press. You're right, this could be improved to happen right after exiting setup.

If setting icons from Home Assistant, the icons will also be downloaded on the next button press (or the next sensor publish in a few minutes). This cannot be improved, because device is in deep sleep between the button presses (or sensor publishes).

Would be great if you could test it with a few new icons just to be sure if it works as I explained above.

daj102, 2023-04-25:
Sure!  I did some testing from HA.  It worked pretty much as you described; I set the labels in HA, then press a button and it says `Downloading icons ...`, refreshes and it shows the icons.

Except ... funnily enough ... the only one that doesn't seem to download is the `mdi:home-assistant` one (it's button 3 in this picture):

[image]

Editing the buttons through the setup page showed the label had been set correctly, but it just didn't download the icon.  I changed it to `mdi:home-automation` through the setup site and that worked AFTER I pressed a button to force it to download.  `mdi:home-assistant` doesn't work AT ALL though, either from setup or HA.

daj102, 2023-04-25:
Oh - just noticed this on the MDI icon page: `This icon has been deprecated and will be removed in an upcoming release` ... hmm, bit naughty that they still show it, but it doesn't download?  Tut.

Playing around with it a bit, there seems to be a little bit a delay in setting the labels; for example, in HA I set buttons 3-6 to a hyphen, clicked off the fields, and pressed the button.  Buttons 3-5 updated, but button 6 stayed as an icon ... I clicked again, and it then changed to a hyphen.  Maybe just a timing delay in MQTT /shrug

Another thought for you: on the Material Design site, there are a couple of other libraries - particularly, the MDI Light set.  I tried putting `mdil:` prefixes in but it didn't take - I know it's a different repo, but they might be better for the eInk display.  Might be a future feature?

And finally - THANK YOU SO MUCH for this awesome little device; it sits on my desk every day and helps me kick off routines and control the music in my overhead speakers, etc.  It's an awesome little gadget, and I really love it.

nplan, 2023-04-26:
As you noticed - deprecated icons are not available.

The delay is probably due to this: you must press enter after inputing the label in the text field or just click outside of it. It's only sent after that.

MDI Light is a much smaller set. I might add it if they expand it in the future.

Thanks, I'm glad you like it. 🙂

closed 2023-04-26 by nplan