nplan/HomeButtons #48: [Feature] Safe Reset Button Combination to prevent accidential resets

gon0, 2023-04-26:
It happens to me 1 to 2 times a month, that I accidentally reset the Homebuttons. Use Cases are:
1. Homebuttons are on the Couch and I accidentally sit on it
2. Homebuttons are mounted on a wall and I accidentally lean on it for too long
Both cases lead to a reset, so that I have to re-configure the Homebuttons.

I would wish for a "Safe Reset Button Combination", which could be activated in the settings for people like me. The default behaviour is just a long-press of any button. When the "Safe Reset Button Combination" is activated, a long-press leads to the "Temperature, Humidity and Battery status" display, but not further (no Setup, no WiFi setup, no reset).

Possible "Safe Reset Button Combinations" could be:
1. Button 1 and Button 6 need to be long-pressed simultaneously, but no other buttons are allowed to be pressed
2. a random or pre-defined combination of the buttons needs to be pressed in sequence. The sequence is indicated by the Button-LEDs. Only if the buttons are pressed in the same sequence, how the LEDs were lit, the setup is unlocked

Probably 1 is the easiest to be implemented from the both scenarios, but I am open for any other solutions.

Keep up the good work!

nplan, 2023-04-26:
Good point. But do I think this should be much better in the v2.1.0. To trigger factory reset, you must now first hold any button for 5 s to open settings. Then release it. And then hold the specific button for 10 s.

I'll think about making the settings menu harder to open.

gon0, 2023-04-26:
Yay! This should already solve my issue, so that a factory reset is not triggered accidentally.

If the setup is accidentally activated, no data is lost. So making it harder to open the settings menu is a nice-to-have for me.

closed 2023-04-26 by gon0