nplan/HomeButtons #49: [Feature] Charging confirmation on screen

manuel-rw, 2023-05-02:
Hello 👋,
first, I would like to thank you for this awesome project. Icons are working great!

Recently, my battery dropped below 40% and I wanted to recharge it. To measure the capacity and current going into the cell, I used a small USB meter to record the data.
I noticed that only little power was drawn and wasn't sure if the charging procedure had been started.

However, after about one minute, it started ramping up to 3.9W. It would be nice to see on the screen, if the device is receiving power and charging.

Maybe an overlay, similar to the ones on Android, would be nice:
[image]

nplan, 2023-05-02:
Hi, thanks!

Charging behaviour depends on the hardware revision: 

**v2.1**
Cannot detect USB connection, but starts charging immediately. It will display a message when charging is done. 

**v2.3**
Detects USB connection, but doesn't wake up when connected. It will only start charging when sensor publish timer triggers or you press a button. At that point it will display a line at the bottom of the screen to indicate that it's charging. When fully charged, the line disappears.

https://docs.home-buttons.com/setup/#charging

I know it's not ideal.

manuel-rw, 2023-05-02:
Ah, I didn't notice that charging indicator.
Since it's full now, I haven't checked if it's actually there or what revision I have.
But perhaps a bigger indicator would still benefit the user, since a small line is barely visible?