nplan/HomeButtons #51: Text size

JunghaJungha, 2023-06-04:
Hi.

I love the new feature of icons in combination with text. But as the text size depends on the length of the text, it makes my display a bit messy.

It would be great if the text size could be the same size independent of the length when the a button label is mixed. 
It actually means the font should be always the smaller font in mixed mode.

Thanks.

nplan, 2023-06-04:
Hi, could you post a picture?

JunghaJungha, 2023-06-06:
[image]
[image]
Please see here both pictures, where I force one to be small by underscore

nplan, 2023-06-29:
Added option to force small font by prepending an underscore in v2.2.1.

closed 2023-06-29 by nplan

JunghaJungha, 2023-06-30:
Thank you, works like a charm