nplan/HomeButtons #53: Toggle Icon

trothe, 2023-06-25:
First of all congrats to this project. This seems to be exactly what I am looking for. One thing though. Is it possible to toggle the icons on the screen. E.g. when you switch the A/C on it would be nice to see the icon inverted or somehow marked as active.

Zixim, 2023-06-26:
that sounds really close to this request : https://github.com/nplan/HomeButtons/issues/34

daj102, 2023-06-26:
I think that the example in #34 gives you an idea of what is possible, @trothe .  I haven't done it myself, but the fact that HA can control the labels is the key.  For example, I have my HomeButtons configured with one button to start my robot vacuum, and one to stop it (they are different service calls).  I could configure an automation to change the label of a single button to start or stop, depending on the status of the vacuum, and then call different services depending on the label.  That would save me a button for other uses ...

It's a lot more work, but I think it's a lot more flexible than specifying on and off icons, tying the button to state to decide which one to show, etc.  I get where you are coming from to be able to bind the button to a state and show different icons - maybe there is a way to write a blueprint for that?

nplan, 2023-06-26:
Having buttons with state (switches) is not supported currently. Though, it is possible to dynamically change the label with a *Home Assistant* automation as @daj102 mentioned.

> If not the question would be why even have a display at all .. I could print a piece of paper and stick it on there.

@trothe The primary function of the device is to act as a trigger for automations. The display is there exactly so you do **not** have to print a piece of paper and stick it there 😀

Adding state to buttons (making it behave as a Switch entity in *Home Assistant*) is tricky because the device is in deep sleep most of the time. If the state were to change externally, the display would show wrong for up to 60 minutes, depending on how the update interval is set by the user.

hilsonp, 2023-07-03:
Some love being sent to #34 ❤️

nplan, 2023-07-11:
Label change by automations / mqtt is improved in v2.3.0. Native toggle / switch support is not planned at the moment.

closed 2023-07-11 by nplan