nplan/HomeButtons #57: Request: Add ability to change URL MDIs are downloaded from

NiccyB, 2023-06-27:
I'd quite like to host the MDIs locally for this and a different project locally. Adding the ability to change the source of the icons would mean my IoT network can remain locked down and all resources are local.

nplan, 2023-06-28:
Good idea. Would you be interested in creating a web server for that? It would convert from SVG to a BMP of desired resolution on the fly. There's already a python script for that here https://github.com/nplan/MDI-BMP

NiccyB, 2023-08-02:
I would love to, but web programming is most definitely not my thing!

Equidamoid, 2024-12-28:
Any way I can help bringing this to life? =)

Not only it removes the internet connection dependency and all the related troubles, but also allows adding custom icons, as MDI is pretty limited.

I can craft the server, but tbh I'm not sure if it's *the* way to do it. Pre-generating the scaled icons and serving them as static files sounds much more reliable. 
As a side feature, your script can be extended to also support icons from thenounproject.com

nplan, 2025-01-04:
@Equidamoid I implemented an option to set a custom icon server. Please try it out and give me feedback.

The server url can be entered on the [setup page](https://docs.home-buttons.com/original/user_guide/#setup) below the label fields.

The server should be a HTTP server serving static BMP files of the correct format and size. For testing https://github.com/nplan/MDI-BMP can be cloned.

Example:
`base_url = http://192.168.1.123/test/icons/`
_HomeButtons_ will expect icons at : `base_url/64x64/alien.bmp`

Folders with sizes 100x100, 64x64 and 48x48 must be present.

_Home Assistant_ can also be used as a server. You can place the icons at `config/www/icons` and set the icon server to `http://192.168.2.2:8123/local/icons/`

The label fields are still in format `mdi:{icon name}`. But for custom icons the `icon name` can by anything.

To go back to the default server (github) just delete the icon server field and save the setup.

[fw_v2.6.0-beta_original.bin.zip](https://github.com/user-attachments/files/18307482/fw_v2.6.0-beta_original.bin.zip)
(unzip first)

Equidamoid, 2025-01-04:
@nplan awesome! will try tomorrow!

nplan, 2025-03-31:
@Equidamoid did you test it? any thoughts?

chennin, 2025-03-31:
It worked for me! Thank you!

Equidamoid, 2025-03-31:
> [@Equidamoid](https://github.com/Equidamoid) did you test it? any thoughts?

Aaaaah, sorry! IRL stuff got in the way.

I just gave it a quick try, and fetching the icons seems to work fine.

Some extra things to be documented:
 - urls: $base_url/64x64/$icon.bmp and $base_url/48x48/$icon.bmp, maybe the format (
 - cache behaviour (e.g. what to do if I updated an icon)

I didn't test https (local net, no need) or homeassistant (don't have it).

nplan, 2025-04-05:
v2.6.0 is now released, documentation about the icon server is [here](https://docs.home-buttons.com/original/user_guide/#icon_server). 

For now, functionality is focused on self hosting. Custom icons work, but are not officially supported.

An icon is saved on device when it's first requested. If the icon file is later changed on the server, the icon on the device will not update. You will have to use a different name.

closed 2025-04-05 by nplan