nplan/HomeButtons #58: Mqqt Error: Mqtt Password Truncated

BriscoeTech, 2023-06-29:
Hello Everyone,

Just received my Home Buttons Mini and was easily able to connect to my wifi network, but ran into a error/bug when trying to setup mqtt.

I have entered in all the settings for my Mosquitto Mqtt Server running in Home Assistant, but I would hit save and the OLED would display "Mqtt Error" and not work.

After trying this multiple times, I noticed the "Mqtt Password" text field only accepts 50 characters, but my randomly generated password in Mosquitto is 64 characters long by default, and the last 14 characters appear to have been lopped off. It looks like the max text box length and maybe memory buffers need to be increased to at least 64 characters so I can enter in the full password.

I have alot of sensors already using my 64 character password, so changing my password to something shorter is not a option.

BriscoeTech, 2023-06-29:
I decided to poke through the code, sure enough, password is 50 characters

[image]

Zixim, 2023-06-29:
mosquitto max password length is 65535 bytes.
UTF-8 characterset can use up to 4 bytes per character... int(65535/4)=16383

nplan, 2023-06-29:
@BriscoeTech I'll raise this limit. In the meantime you could add a new user for MQTT with a shorter password.

@Zixim I can't store this much data 🙂. 50 is already very long for a password... What would be a reasonable limit? 128 characters?

Zixim, 2023-06-29:
@nplan i feel 50 is sufficient... 😛 but I also understand OP's pain of having to redo all the passwords.
64 or 128 are pretty numbers i guess.

nplan, 2023-06-29:
Increased to 64 character in v2.2.1.

closed 2023-06-29 by nplan

BriscoeTech, 2023-07-02:
Thanks everyone! I will take a look at setting up the multiple password pairs.

I agree 64 is probably plenty big enough, I was amazed Mosquito generated that large a password to begin with, and did not think about the implications at the time. Hopefully this will help prevent another user from getting the same surprise.