nplan/HomeButtons #62: Update failed

FLeXO21, 2023-08-30:
After updating my HomeButtons the device was unable to boot.

The only firmware that seems to boot is this one: '2301-112_v2.0.4_full_image.bin''
My device S/N is 2301-048 so this firmware is not exactly for my device since the MAC adres is different.

From the debug port i get the following:

ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6100,len:0x11b0
load:0x4004c000,len:0x8b4
load:0x40050000,len:0x2d88
entry 0x4004c198
[   782][I][main.cpp:959] setup(): [DEVICE] starting...
[   782][I][main.cpp:249] main_task(): [DEVICE] woke up.
[   782][I][main.cpp:250] main_task(): [DEVICE] cpu freq: 240 MHz
[   787][I][main.cpp:251] main_task(): [DEVICE] SW version: v2.0.6
[   801][I][main.cpp:270] main_task(): [HW] version: 2.3
[   802][I][hardware.cpp:29] init(): configured for hw version: 2.3
[   834][I][display.cpp:50] begin(): [DISP] begin
[   837][I][main.cpp:296] main_task(): [DEVICE] batt present: 0, DC connected: 1
[   838][I][main.cpp:355] main_task(): [DEVICE] usr awake mode: 0, awake mode: 0
[   842][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=10 scl=11 freq=100000
[   852][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[   875][I][main.cpp:384] main_task(): [DEVICE] wakeup cause: RESET
[  7152][I][display.cpp:69] update(): [DISP] ended.
[  7158][I][main.cpp:53] start_esp_sleep(): [DEVICE] deep sleep... z z z
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6100,len:0x11b0
load:0x4004c000,len:0x8b4
load:0x40050000,len:0x2d88
entry 0x4004c198
[   513][I][main.cpp:959] setup(): [DEVICE] starting...
[   513][I][main.cpp:249] main_task(): [DEVICE] woke up.
[   513][I][main.cpp:250] main_task(): [DEVICE] cpu freq: 240 MHz
[   518][I][main.cpp:251] main_task(): [DEVICE] SW version: v2.0.6
[   532][I][main.cpp:270] main_task(): [HW] version: 2.3
[   533][I][hardware.cpp:29] init(): configured for hw version: 2.3
[   565][I][display.cpp:50] begin(): [DISP] begin
[   568][I][main.cpp:296] main_task(): [DEVICE] batt present: 0, DC connected: 1
[   569][I][main.cpp:355] main_task(): [DEVICE] usr awake mode: 0, awake mode: 0
[   573][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=10 scl=11 freq=100000
[   583][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[   606][I][main.cpp:370] main_task(): [DEVICE] wakeup cause: PIN 5
E (6673) phy_init: load_cal_data_from_nvs_handle: calibration data MAC check failed: expected 58:cf:79:b3:94:5c, found 58:cf:79:b3:94:76

How to go from here, i'm unable to flash any other image

Zixim, 2023-08-30:
have you tried the [full image](https://docs.home-buttons.com/original/update/#full_image) flash ?

FLeXO21, 2023-08-30:
> have you tried the [full image](https://docs.home-buttons.com/original/update/#full_image) flash ?

Yes i have flashed v2.4.0 full:
esptool.exe --port COM4 --after no_reset write_flash 0x0 fw_v2.4.0_original_full_image.bin

The device gives a HW init problem after unable to read the preferences?

ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6100,len:0x129c
load:0x4004c000,len:0xaf4
load:0x40050000,len:0x2f2c
entry 0x4004c1b8
[   353][I][APP     ] starting...
[   354][I][APP     ] woke up.
[   354][I][APP     ] cpu freq: 240 MHz
[   354][I][APP     ] SW version: v2.4.0
[   356][I][HW      ] migrating factory params from nvs to efuse
[   811][E][Preferences.cpp:50] begin(): nvs_open failed: NOT_FOUND
[   368][E][HW      ] failed to read factory params from nvs
[   372][I][HW      ] read efuse factory params: SN=, RID=, M=, HW=
[   378][E][HW      ] factory params empty
[   382][E][APP     ] HW init failed! Going to sleep.
E (388) sleep: Not an RTC IO: GPIO22
[   392][I][APP     ] deep sleep... z z z

If i flash fw_v2.3.0:
>esptool.exe --port COM4 --after no_reset write_flash 0x0 fw_v2.3.0_original.bin
I get this from the console:

invalid header: 0x746165▒ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
invalid header: 0x74616572
invalid header: 0x74616572
invalid header: 0x74616572
invalid header: 0x74616572
invalid header: 0x74616572
invalid header: 0x74616572
invalid header: 0x74616572
invalid header: 0x74616572
invalid header: 0x74616572
invalid header: 0x74616572
invalid header: 0x74616572
invalid header: 0x74616572
invalid header: 0x74616572
invalid header: 0x74616572
etc

FLeXO21, 2023-08-30:
The firmware from 2301-048 was a last resort, and is the only one that booted
All the other firmware's reported the invalid header messages prior to flashing the 2301-048 firmware

FLeXO21, 2023-08-30:
> @FLeXO21 what fw version did you try to update from?

from this version: fw_v2.3.0_original.bin

nplan, 2023-08-30:
Versions 2.0.6 and earlier have factory parameters saved to flash. If you overwrite this section of the flash by flashing the firmware image at the wrong offset, the device won't boot. This is fixed in v2.1.0 and later by transferring factory data to hardware eFuses.

Here are the steps to fix your device:
 1. Connect device with an USB-C cable to your computer
2. Go to https://nplan.github.io/HomeButtonsFlasher/
3. Put device to programming mode by holding BOOT button and pressing RST button. Boot LED should light up.
4. Select **Home Buttons**, **V2.4.0** and **Full Image**
5. Click Connect, select ESP32-S2 from the list and click Connect
6. Click Install V2.4.0 ORIGINAL FULL IMAGE
7. Check Erase device and click Next
8. Confirm installation
9. Wait about 2 minutes for installation to complete
10. Close the browser tab and go to https://adafruit.github.io/Adafruit_WebSerial_ESPTool/
11. On device hold BOOT button and press RST button. Boot LED should light up.
12. Select 921600 Baud and click Connect
13. Select ESP32-S2 from the list & click Connect
14. In the first offset field enter 9000 and choose the provided **NVS bin file**
15. Click Program and wait a few seconds
16. On device press RST button. Home Buttons will show welcome screen.
17. Set up the device again according to the guide https://docs.home-buttons.com/original/setup/

After this procedure, the factory data is transferred to hardware eFuses. Erasing flash will not brick the device anymore, but it has to be programmed with the **full image** firmware afterwards.

Please send an email with your serial number to **support@home-buttons.com**, so I can send you the correct NVS bin file. I don't want to upload it here, since I don't want other users flashing it. It's specific to the device serial number.

FLeXO21, 2023-08-30:
Wonderful! Thank you for providing such detailed information. I will make sure to follow the procedure you've outlined.