nplan/HomeButtons #64: Icons not displaying

antonpiatek, 2023-09-13:
Thanks for building these, am looking forward to getting it all setup and working.

I'm configuring directly via MQTT, publishing the btn label with value `mdi:play-pause` or `mdi:lightbulb-auto-outline`. The device is in online mode for now so as soon as I publish the label it goes into `downloading icons` and then shows the screen with the label as the `mdi:file-question-outline` icon.

Is there any logging via serial I can setup to get an idea of why the download perhaps failed?

nplan, 2023-09-14:
If there was an internet connection issue, the display would show `Icon server not reachable`.

I would check the MQTT messages first. Could you share how you are publishing them?

Logging is possible via USB to Serial adapter. You can connect it to the pins on the PCB next to the RST button. RX of the adapter goes to TX on Home Buttons, and GND to GND. The level should be set at 3.3V and the baud is 115200.

antonpiatek, 2023-09-16:
Working now - I think I had a local DNS issue (wasn't obvious as only my Chromecast had issues, everything else was working).

Having version number on an api would still be nice though

closed 2023-09-16 by antonpiatek