nplan/HomeButtons #67: [Bug] Using '{% ... %}' syntax in label bricks device

mrvdot, 2023-10-04:
I was setting up my HB (Original) and was attempting to make one of the labels show a counter (either solo or ideally next to an icon).

I tried this by setting the label (via HomeAssistant) to `{% states('input_number.daily_water') %}`. Upon next update, the whole device got stuck at "RESTART" screen, and just repeatedly blinked. I tried removing the battery, holding RST, etc. Eventually I flashed a full new image and started again (thinking it was a fluke).

This time, I tried to use `set_value` with same string to update the label*. Exact same problem :disappointed: 

---

**Firmware Version**: 2.4.0 Original
**Hardware Version**: 2.4

---

* **Note**: I am (now) aware that templates aren't supported in Device Actions and I need to use `number.set_value` or `text.set_value` instead. That said, I don't think the device should brick just because someone tries :)

nplan, 2023-10-08:
I can replicate it. Thanks for reporting, will investigate further.

nplan, 2023-10-30:
Fixed in v2.4.1.

closed 2023-10-30 by nplan