nplan/HomeButtons #70: Automatic WiFi reconnection

chriszuercher, 2023-11-14:
I discovered an issue using the latest version of HomeButtons. My WiFi is shutdown during night. This leads to a "Network error" message on HomeButtons every morning. A button press get it working again. Would it be possible to automatically reconnect to WiFi from time to time to check if a connection recovered? Eg. ever 30 minutes? This would be perfect.

Thanks for this great product!

nplan, 2023-11-15:
This is intended behaviour. Trying to connect to Wi-Fi drains a lot of energy. To preserve battery, it stops after a couple tries. Even trying every 30 minutes would significantly reduce the expected battery life.

chriszuercher, 2023-11-15:
Could this be disabled as an option so users can define it? It uses more energy to have the wifi running all night in comparison to charging the battery more often 😉

chriszuercher, 2024-09-19:
@nplan is there any chance this sill be implemted in future?

Or can I do it myself by modifing the code and deploy it to my device using USB? Is there anything to consider? i would try to add a DeepSleep wakeup in a defined internal (e.g 60min) to check the connection and try to reconnect it. If possible added as setting.