nplan/HomeButtons #71: After Update v2.4.1: single button pressed automatisation are not executed anymore / reliably transmitted (?)

no2spc, 2023-12-03:
I did an upgrade from v2.1 to 2.3.1v OTA - followed via 'Home Buttons Flasher' a full image to v2.4.1 on a Hardware Revision 2.1. Set up wifi and credentials for MQTT, and the home buttons is recognised in HA (showing device infos, reading temperature, battery status and humidity, and also 'Show Message' works when a button is pressed. However, the single press automatisations are no executed/called anymore, the double/tripple click work fine. I checked that the trigger, device / button ID are correct and also started with blank new automatisation: according to HA they are never executed, and if I interpret thev MQTT log, they are also not recieved. 

MQTT Log, pressing first button 1 once, 20 s later followed by button 1 pressed twice:

```
2023-12-03 10:22:13.962 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/awake_mode/available (qos=0): b'offline'
2023-12-03 10:22:13.962 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/sensor_interval (qos=0): b'10'
2023-12-03 10:22:14.005 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_1_label (qos=0): b'mdi:lightbulb'
2023-12-03 10:22:14.005 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_2_label (qos=0): b'mdi:television'
2023-12-03 10:22:14.005 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_3_label (qos=0): b'mdi:heart-multiple'
2023-12-03 10:22:14.006 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_4_label (qos=0): b'mdi:rocket-launch'
2023-12-03 10:22:14.008 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_5_label (qos=0): b'mdi:weather-night'
2023-12-03 10:22:14.008 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_6_label (qos=0): b'mdi:sofa-single'
2023-12-03 10:22:14.009 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/awake_mode (qos=0): b'OFF'
2023-12-03 10:22:14.009 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/disp_msg (qos=0): b'-'
2023-12-03 10:22:38.784 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/awake_mode/available (qos=0): b'offline'
2023-12-03 10:22:38.831 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/sensor_interval (qos=0): b'10'
2023-12-03 10:22:38.832 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_1_label (qos=0): b'mdi:lightbulb'
2023-12-03 10:22:38.832 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_2_label (qos=0): b'mdi:television'
2023-12-03 10:22:38.832 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_3_label (qos=0): b'mdi:heart-multiple'
2023-12-03 10:22:38.833 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_4_label (qos=0): b'mdi:rocket-launch'
2023-12-03 10:22:38.833 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_5_label (qos=0): b'mdi:weather-night'
2023-12-03 10:22:38.833 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_6_label (qos=0): b'mdi:sofa-single'
2023-12-03 10:22:38.834 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/awake_mode (qos=0): b'OFF'
2023-12-03 10:22:38.835 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/disp_msg (qos=0): b'-'
2023-12-03 10:22:39.321 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/button_1_double (qos=0): b'PRESS'
2023-12-03 10:22:39.326 DEBUG (MainThread) [homeassistant.components.mqtt.client] Transmitting retained message on homebuttons/HomeButtons/cmd/btn_1_label: 'mdi:lightbulb', mid: 24, qos: 0
2023-12-03 10:22:39.330 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/temperature (qos=0): b'20.39'
2023-12-03 10:22:39.386 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/humidity (qos=0): b'52.90'
2023-12-03 10:22:39.388 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/battery (qos=0): b'84'
2023-12-03 10:22:39.388 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_1_label (qos=0): b'mdi:lightbulb'
``` 

Any ideas / leads where to look for the culprit?

nplan, 2023-12-03:
Please try to restart Home Buttons by opening the [Settings menu](https://docs.home-buttons.com/original/user_guide/#settings) (hold 2 buttons together for 5 seconds) and click the restart button.

After it restarts, press any button. It will re-send the MQTT discovery config messages. This should fix the issue.

no2spc, 2023-12-03:
Thanks for the reply; I restarted the device and pushed a button. Adapted the routines, unfortunately it still doesn't work. Button 2: pressed once, pressed double, pressed once.

```
2023-12-03 19:57:24.311 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/awake_mode/available (qos=0): b'offline'
2023-12-03 19:57:24.312 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/sensor_interval (qos=0): b'10'
2023-12-03 19:57:24.313 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_1_label (qos=0): b'mdi:lightbulb'
2023-12-03 19:57:24.373 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_2_label (qos=0): b'mdi:television'
2023-12-03 19:57:24.374 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_3_label (qos=0): b'mdi:heart-multiple'
2023-12-03 19:57:24.374 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_4_label (qos=0): b'mdi:rocket-launch'
2023-12-03 19:57:24.375 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_5_label (qos=0): b'mdi:weather-night'
2023-12-03 19:57:24.376 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_6_label (qos=0): b'mdi:sofa-single'
2023-12-03 19:57:24.376 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/awake_mode (qos=0): b'OFF'
2023-12-03 19:57:24.377 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/disp_msg (qos=0): b'-'
2023-12-03 19:57:24.612 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/temperature (qos=0): b'20.09'
2023-12-03 19:57:24.617 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/humidity (qos=0): b'59.04'
2023-12-03 19:57:24.618 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/battery (qos=0): b'84'
2023-12-03 19:57:37.486 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/awake_mode/available (qos=0): b'offline'
2023-12-03 19:57:37.550 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/sensor_interval (qos=0): b'10'
2023-12-03 19:57:37.551 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_1_label (qos=0): b'mdi:lightbulb'
2023-12-03 19:57:37.552 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_2_label (qos=0): b'mdi:television'
2023-12-03 19:57:37.552 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_3_label (qos=0): b'mdi:heart-multiple'
2023-12-03 19:57:37.553 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_4_label (qos=0): b'mdi:rocket-launch'
2023-12-03 19:57:37.553 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_5_label (qos=0): b'mdi:weather-night'
2023-12-03 19:57:37.554 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_6_label (qos=0): b'mdi:sofa-single'
2023-12-03 19:57:37.554 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/awake_mode (qos=0): b'OFF'
2023-12-03 19:57:37.554 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/disp_msg (qos=0): b'-'
2023-12-03 19:57:38.115 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/button_2_double (qos=0): b'PRESS'
2023-12-03 19:57:38.134 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/temperature (qos=0): b'20.11'
2023-12-03 19:57:38.137 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/humidity (qos=0): b'59.59'
2023-12-03 19:57:38.137 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/battery (qos=0): b'83'
2023-12-03 19:57:38.345 DEBUG (MainThread) [homeassistant.components.mqtt.client] Transmitting retained message on homebuttons/HomeButtons/cmd/btn_2_label: 'mdi:television-off', mid: 80, qos: 0
2023-12-03 19:57:38.352 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_2_label (qos=0): b'mdi:television-off'
2023-12-03 19:57:47.672 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/awake_mode/available (qos=0): b'offline'
2023-12-03 19:57:47.673 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/sensor_interval (qos=0): b'10'
2023-12-03 19:57:47.718 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/btn_1_label (qos=0): b'mdi:lightbulb'
[... cut ...]
2023-12-03 19:57:47.723 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/disp_msg (qos=0): b'-'
2023-12-03 19:57:47.988 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/temperature (qos=0): b'20.12'
2023-12-03 19:57:48.006 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/humidity (qos=0): b'60.15'
2023-12-03 19:57:48.008 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homebuttons/HomeButtons/battery (qos=0): b'83'
```

closed 2023-12-03 by no2spc

nplan, 2023-12-03:
Could you please use [MQTT Explorer](https://mqtt-explorer.com/) to see if messages appear on the MQTT topics.

nplan, 2023-12-04:
Glad to hear you sorted it out.

The *button1* etc. without the underline are topics from the very old software version v1.0.0. I have no idea why it has become the issue now...

MQTT discovery in Home Assistant is not working perfectly all the time. Normally, manually deleting all discovery config topics in MQTT Explorer and then resending the config messages from Home Buttons fixes this kind of issues.