nplan/HomeButtons #73: home button controls do not show up in home assistant

RMHogervorst, 2024-01-06:
Hi there!
Love this product! 

1. I activated home button (mini), 
2. connected it to the wifi, 
3. found the the settings page
4. set up the MQTT details
5. The device is discovered in home assistant

However, I only see 
Sensors: (Battery, Humidity, Temperature)
Controls: sensor interval
Logbook is empty

I have firmware 2.4.1, hardware 1.1

When I click on the MQTT info button (in home button) in home assistant I do see the other values, but is it possible that HA is not interpreting this right?

```
Transmitted messages:

Home Buttons 342095 Button 1 label (text.home_buttons_342095_button_1_label)
MQTT discovery data:

    Topic: homeassistant/text/HBTNS-2305-233-342095/button_1_label/config 

```

payload:
```
name: Button 1 label
unique_id: HBTNS-2305-233-342095_button_1_label
command_topic: homebuttons/Home Buttons 342095/cmd/btn_1_label
state_topic: homebuttons/Home Buttons 342095/btn_1_label
max: 56
icon: mdi:numeric-1-box
retain: 'true'
device:
  identifiers:
    - HBTNS-2305-233-342095
platform: mqtt
```

Subscribed topics:

    homebuttons/Home Buttons 342095/btn_1_label

RMHogervorst, 2024-01-06:
whoops, the buttons are disabled under 'Entities'

nplan, 2024-01-06:
Were you able to enable the entities? Is it working now?

RMHogervorst, 2024-01-06:
I think they came alive after a while, it all works!

closed 2024-01-06 by RMHogervorst