nplan/HomeButtons #74: Single press does not work anymore with Home Assistant

ajfriesen, 2024-02-13:
I am on firmware `2.4.1`.
Hardware: `2.1`
And since a few days, maybe with the latest Home Assistant update the buttons does not work as before.

I had this automation configured which worked fine but does not work at all anymore.

```yaml
alias: Fernbedienung Stern Schalten
description: Schalte Stern ein und aus
trigger:
  - platform: device
    domain: mqtt
    device_id: a8ea862498ed0c28dade6374f1b1a9d9
    type: button_short_press
    subtype: button_5
condition: []
action:
  - service: light.toggle
    data: {}
    target:
      entity_id: light.sterne
mode: single
```

I checked and none of the buttons work anymore with a single press even though I see the led flash.

So I tried a double press and who knew it just works:

```yaml
alias: Fernbedienung Stern Schalten
description: Schalte Stern ein und aus
trigger:
  - platform: device
    domain: mqtt
    device_id: a8ea862498ed0c28dade6374f1b1a9d9
    type: button_double_press
    subtype: button_5
condition: []
action:
  - service: light.toggle
    data: {}
    target:
      entity_id: light.sterne
mode: single

```

Any idea what I could have done wrong that a double press works but a single press does not?

nplan, 2024-02-13:
It that doesn't fix it, please take a look at issue #71

ajfriesen, 2024-02-13:
Thanks.
Encountered the #71  a while ago now and fixed it the same way.

Now I had to reset and then I had to redo the automation.
choosing a different entity and then use the correct entity. Only that way I could restore 1 button press.

Weird.

closed 2024-02-13 by ajfriesen