nplan/HomeButtons #75: Please reset IP-Configuration after WiFi-Setup ?

Ventusfahrer, 2024-02-16:
Hi, 

Situation:

I was preparing a HomeButton for a friend of mine. For this I connected the HB to my local network and was accessing his HA-addon based mqtt via internet.  I was able to finish development of the Node-Red based automations and need to transfer the device to him.

At his site I wanted to connect the HB to his WLAN, where I failed.

Why?
Well, its more or less curious: 
I brought the device into configuration mode (2 buttons pressed, until the config mode shows up), the I select the WiFi symbol. The device was the connected using it's AP
After that I provided the new WiFi credentials (SSID + password) and the device was rebooting, 
But I was never able to connect it in his network.

Reason:
I typically configure the IoT devices using fixed IP-Adresses. There are several reasons to do it this way, connection performance and energy-consumption is one of them, timing issues while initializing other hardware-components (displays for example) when DHCP is just handshaking with the DHCP server for getting the IP-Address.

My ipv4 subnet looks like 192.168.221.xxx, his one is 192.168.178.xxx.

So, when I only change SSID and password, the device is connecting to the new WiFi, it is performing something like a DHCP connection (I could see it's new DHCP IP-Address in the WiFi-Router), but after the connection is set up, the device changes it's IP-Configuration to the old static IP-Address. As a result we have a device configured with an IP-Address, Subnet-Mask etc. which does not fit the network it is connected to. This will never work.

As a result I need to perform a factory reset, but it takes a while until I did unterstand what's going on.

To get around this issues you should always erase existing IP-Configurations whenever WiFi-Settings are changed.

Beste regards

Zixim, 2024-02-16:
That's just how network configuration works. If you configure the device without DHCP, and you then put it in a network that doesn't match your fixed ip settings... you just shot yourself in the foot.
Every device that has networking, functions in this way.

Ventusfahrer, 2024-02-16:
well, I partly agree.

you typically have all WiFi relevant information in one page/form. 
First the wifi connection parameters, then a checkbox whether DHCP or not, and if not using DHCP the rest of the parameters. 

As soon you seperate this information on different pages which automaticaly performd reboot when saving the WiFi changes the possible upcoming problems are not visible to you at all.

After saving the information you will be very surprised and a lot of people with less network centric background will be lossed.

nplan, 2024-10-13:
Fixed in v2.5.1. Static IP config now resets when changing Wi-Fi network.

closed 2024-10-13 by nplan