nplan/HomeButtons #76: Icons not downloading

micro1995, 2024-03-21:
When using any icon, on the next button press, my HomeButton Mini says "Downloading Icons" on the Display, but the Icon is not available afterwards.
When looking at the log I think the problem are the following lines:

[  3876][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():273]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[  3879][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -9984
[  3444][E][Download] GET request failed with code -1
[  3448][W][APP     ] icon server NOT reachable

Network is connected, MQTT works, IP Configuration via DHCP.

Is there a problem with the Certificate used in the source code?

nplan, 2024-03-21:
It appears to be a problem with the certificate, I'm seeing the same error on my test devices. I'll investigate further...

nplan, 2024-03-24:
fixed in v2.4.2

closed 2024-03-24 by nplan

jiserra, 2024-08-12:
I updated to 2.4.3 today (I was in 2.3.0) and the icons are not downloading for me. I did the troubleshooting but they still appear like the icon is missing. Is there anything else I can try? I have the original 6 button device (Model A1 rev 2.3)

jiserra, 2024-08-12:
Ok, it seems that I wasn't triggering the "Downloading icons" in a good way… it's working now!