nplan/HomeButtons #80: Icons not downloading

chennin, 2025-03-31:
Hello! I reset my original HomeButton (purposefully) as I was updating the firmware (2.5.1 seems to send more reliable than 2.4.0, so that's nice), and now it can't download icons. The devices flashes "Downloading Icons" but only comes back with the file-question-outline icon.

[image]

Possibly similar to #76? It looks like the certificate for `raw.githubusercontent.com` no longer chains to Digicert R2.

Is there a way to get to the logs without a USB to UART adapter? I have Linux, and Chrome for any ESP Web tools.

[image]

nplan, 2025-03-31:
Hi! I see the same issue. It's the certificate. Will have to be fixed by an update. In the meantime you can test the beta firmware from [here](https://github.com/nplan/HomeButtons/issues/57#issuecomment-2571351837) with the ability to host icons locally by Home Assistant.

nplan, 2025-04-05:
Fixed in v2.6.0.

closed 2025-04-05 by nplan